openrocket/openrocket #1518: Feature request: don't require preset to be checked to be applied

SiboVG, 2022-07-04:
Right now, the only way to apply a preset is to first check the preset in the preset dialog, and then to either select the (checked) preset in the preset dialog, or to close the preset dialog and then select the preset from the preset drop down menu in the component edit dialog.

I propose to also apply a preset when it is highlighted in the preset dialog, without being checked.

The only thing that makes this issue not an easy feat is: how would you display a preset in the preset combo box that is unchecked (so not a favorite) vs one that is checked (a favorite). Maybe to make it clear that checked presets are seen as 'favorites', you could add a star icon next to the preset name? I'm open to other suggestions.
[image]

hcraigmiller, 2022-07-04:
Use a star icon and order the presets first, followed by recently used (no stars).

hcraigmiller, 2022-07-04:
I think @neilweinstock prefers this order:

[image]

neilweinstock, 2022-07-05:
> Right now, the only way to apply a preset is to first check the preset in the preset dialog, and then to either select the (checked) preset in the preset dialog, or to close the preset dialog and then select the preset from the preset drop down menu in the component edit dialog.

Since when is this the case?  It has never been so, unless a PR snuck in that made this change while I wasn't looking.

SiboVG, 2022-07-11:
> Since when is this the case? It has never been so, unless a PR snuck in that made this change while I wasn't looking. I don't even want to look at the proposed solutions until I'm clear on why this is a problem in the first place.

Whoops, sorry @neilweinstock, I was wrong, this is still the case (thought I tested this before, but I guess not thoroughly enough). But even though the initial reason for this issue is wrong, maybe it is useful to add a distinction between favorites and recently used presets (as proposed by @hcraigmiller)?

neilweinstock, 2022-07-11:
> Whoops, sorry @neilweinstock, I was wrong, this is still the case (thought I tested this before, but I guess not thoroughly enough). But even though the initial reason for this issue is wrong, maybe [...]

I originally proposed the Recently Used function, and after further thought decided that it's more trouble than it's worth (discussed in another PR or issue somewhere).  But either way, it's completely unrelated to the subject of this feature request.

In fact, since the premise on which this feature request was in error, I think this issue should just be closed.  Discussion of various ways to improve and/or rethink the way presets work should continue elsewhere.

commit 452c6e22 (2022-07-19): Fix #1518
closed 2022-07-11 by SiboVG