openrocket/openrocket #1519: Don't update sim connected to current configuration if option not checked

neilweinstock, 2022-07-05:
Another one that stems from this report on TRF: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-4-is-now-available.171295/post-2294520.  This is a little obscure and weird but I think it should be fixed.

The problem arises if the user has "Update estimated flight parameters in design window" CHECKED, but also has "run outdated simulations when you open the simulation tab" UNCHECKED.  In this case, the flight parameters shown in the design window are connected to a specific simulation, and when you go to the simulation tab, that simulation is shown green (up-to-date) because it was updated as part of the design tab updates.  This is counter to the user's expectation, which is that going back to the simulation tab should show all out-of-date simulations.

The high level hand-wavey fix for this is presumably to create an "invisible" simulation that is used for the flight parameters in the design window, completely separate from any of the sims in the simulation tab.

hcraigmiller, 2022-07-05:
Fixed, see closed PR #1510.

hcraigmiller, 2022-07-05:
I should have been more specific. The complaint of the user you refer to has been fixed, this describes the [new functionality](https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-4-is-now-available.171295/page-18#post-2294708).

The new functionality (current unstable), as applied to checking "Update estimated flight parameters in design window" and unchecking "run outdated simulations when you open the simulation tab" is premised on this:

1. If a _Flight configuration_ is selected, then the "control" simulation (the first simulation using that _Flight configuration_) will always be re-run when the rocket configuration is changed, updating the flight parameters in design window.
2. If the _Flight configuration_ is set to _No motors_, then no simulations will be re-run when the rocket configuration is changed, and the flight parameters will always be "N/A."

By way of example:

[image]

I believe that this is the correct functionality for these settings.

neilweinstock, 2022-07-05:
What I am saying is that the new functionality does not address the user's complaint, *and* it's really still not behavior that I call internally self-consistent.  The "Update estimated flight parameters in design window" option does not in any way imply that a simulation in the sim tab will get updated.   I am saying that the flight parameters in the design window should be completely separate from any sim in the sim tab, even if the design window sim is basically a clone of one of the existing simulations.

hcraigmiller, 2022-07-05:
I sent the user a .jar, and he hasn't said that it didn't fix his complainet. . . he that us all for the fix.

hcraigmiller, 2022-07-05:
> I am saying that the flight parameters in the design window should be completely separate from any sim in the sim tab, even if the design window sim is basically a clone of one of the existing simulations.

If you are suggesting that the flight parameters in the design window should be **completely disconnected** from any simulation, how are the estimates to be updated as changes are made?

neilweinstock, 2022-07-05:
> If you are suggesting that the flight parameters in the design window should be **completely disconnected** from any simulation, how are the estimates to be updated as changes are made?

An internal clone of a simulation is made and used for the design window.

> I sent the user a .jar, and he hasn't said that it didn't fix his complainet. . . he thanked us all for the fix.

That is good but I still think the way things work right now is not really correct.  I fully acknowledge, though, that it is fairly esoteric at this point and maybe not worth pursuing.

hcraigmiller, 2022-07-05:
> An internal clone of a simulation is made and used for the design window.

And what launch conditions and simulation options would be used?

hcraigmiller, 2022-07-05:
> A clone of an existing simulation. I don't know how else to say it.

That's what I'm trying to understand. Currently, the estimated flight parameters in the design window are a clone of the "control" for the selected _Flight configuration_.

Is this what you are suggesting, that when the "Update estimated flight parameters in design window" is **checked**, but the "Run outdated simulations when you open the simulation tab" is **unchecked,** that **none of the simulations update** when a rocket configuration is changed, but **the estimated flight parameters in design window do update**?

neilweinstock, 2022-07-05:
I expect it to do what the options say it will do, not what is convenient for the developers:

1) update the flight parameters in the design window
2) *don't* update any of the simulations in the sim tab

Again, if no one cares than I'm willing to drop this.  We've got bigger fish to fry right now.  It's conceivable that this issue will come up down the road, connected to something else, but if so we can deal with it then.

hcraigmiller, 2022-07-05:
Again, just trying to make sure that I understand your suggestion so that I can evaluate and appropriately comment.

I disagree with your proposal for several reasons.
- My recollection is that part of the reason that the current functionality exists is because users complained that there was a disconnect between what appeared on the estimated flight parameters in the design window and the simulation values (users wanted to know why the values didn't always match). This resurrects that complaint, albeit in a slightly different context.
- If a user truly does not want any of the simulations updated, setting the _Flight configuration_ to _No motors_ achieves that result.
-  The estimated flight parameters in the design window do not reflect all of the flight parameters that users may be attempting to control for, such as velocity at deployment or ground hit velocity, and this change would take away the flexibility of being able to dynamically watch how rocket configuration changes affect such parameters on the _Flight simulations_ pane.
- Allowing a label to control functionality as an application evolved can cause development stagnation. If the basis of your reasoning is nothing more that "[you} expect it to do what the options say it will do. . ." perhaps the label should be changed to something along the lines of "Run _inactive_ out-dated _flight configuration_ simulations when rocket configurations are changed" (this better describes the functionality). A label should not be given priority over flexibility and functionality.

My two cents.

neilweinstock, 2022-07-05:
Rather than continue on quibbling about this, I'm just going to close it, and again we can revisit in the future if for some reason it comes up again.

closed 2022-07-05 by neilweinstock