openrocket/openrocket #1520: PNG crashes scale fin

jeffreyg-sh, 2022-07-07:
---------- Bug report ----------  Please include a description about what actions you were performing when the exception occurred: (You can edit text directly in this window) 
 1. Imported image (attached)
 2. Click Scale  
 3. Enjoy Stack trace. Include your email address (optional; it helps if we can contact you in case we need additional information):     (Do not modify anything below this line.) 
 ---------- Exception stack trace ---------- java.lang.IllegalArgumentException: setSelectedIndex: 1 out of bounds at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622) at net.sf.openrocket.gui.dialogs.ScaleDialog.init(ScaleDialog.java:358) at net.sf.openrocket.gui.dialogs.ScaleDialog.<init>(ScaleDialog.java:235) at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$3.actionPerformed(FreeformFinSetConfig.java:245) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) 
 ---------- Thread information ---------- 
Thread[AWT-EventQueue-0,6,main]  
---------- System information ---------- OpenRocket version: 22.02.beta.04 OpenRocket source: default OpenRocket location: /Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/app/OpenRocket.jar JOGL version: 2.4.0-rc-20210111 Current default locale: en_US System properties: apple.laf.useScreenMenuBar=true awt.toolkit=sun.lwawt.macosx.LWCToolkit com.apple.mrj.application.apple.menu.about.name=OpenRocket exe4j.moduleName=/Applications/OpenRocket 22.02.beta.04.app file.encoding=UTF-8 file.separator=/ ftp.nonProxyHosts=local|*.local|169.254/16|*.169.254/16 gopherProxySet=false http.nonProxyHosts=local|*.local|169.254/16|*.169.254/16 i4j.jreBundle=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/jre.bundle i4j.ownBundlePath=/Applications/OpenRocket 22.02.beta.04.app install4j.appDir=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/app/ install4j.exeDir=/Applications/ install4j.launcherId=59 install4j.swt=false java.awt.graphicsenv=sun.awt.CGraphicsEnvironment java.awt.printerjob=sun.lwawt.macosx.CPrinterJob java.class.path=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/app/.install4j/i4jruntime.jar:/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/app/OpenRocket.jar java.class.version=55.0 java.home=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/jre.bundle/Contents/Home java.io.tmpdir=/var/folders/h5/l_85cxhx0y97z5d00ksrbsdc0000gn/T/ java.library.path=/Users/jgortatowsky/Library/Java/Extensions:/Library/Java/Extensions:/Network/Library/Java/Extensions:/System/Library/Java/Extensions:/usr/lib/java:. java.runtime.name=OpenJDK Runtime Environment java.runtime.version=11.0.15+10-LTS java.specification.name=Java Platform API Specification java.specification.vendor=Oracle Corporation java.specification.version=11 java.util.Arrays.useLegacyMergeSort=true java.vendor=BellSoft java.vendor.url=https://bell-sw.com/ java.vendor.url.bug=https://bell-sw.com/support java.version=11.0.15 java.version.date=2022-04-19 java.vm.compressedOopsMode=Zero based java.vm.info=mixed mode java.vm.name=OpenJDK 64-Bit Server VM java.vm.specification.name=Java Virtual Machine Specification java.vm.specification.vendor=Oracle Corporation java.vm.specification.version=11 java.vm.vendor=BellSoft java.vm.version=11.0.15+10-LTS jdk.debug=release jnlp.jogamp.tmp.cache.root=jln4241109311750523683 line.separator=\u000a os.arch=x86_64 os.name=Mac OS X os.version=11.6.2 path.separator=: socksNonProxyHosts=local|*.local|169.254/16|*.169.254/16 sun.arch.data.model=64 sun.awt.enableExtraMouseButtons=true sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler sun.boot.library.path=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/jre.bundle/Contents/Home/lib sun.cpu.endian=little sun.cpu.isalist= sun.font.fontmanager=sun.font.CFontManager sun.io.unicode.encoding=UnicodeBig sun.java.command=com.install4j.runtime.launcher.MacLauncher sun.java.launcher=SUN_STANDARD sun.jnu.encoding=UTF-8 sun.management.compiler=HotSpot 64-Bit Tiered Compilers sun.os.patch.level=unknown user.country=US user.dir=/Applications/OpenRocket 22.02.beta.04.app/Contents/Resources/app user.home=/Users/jgortatowsky user.language=en user.name=jeff user.timezone=America/Los_Angeles ---------- Error log ---------- 460 23.008 ERROR (SwingExceptionHandler.java:48) Handling uncaught exception on thread=Thread[AWT-EventQueue-0,6,main] > java.lang.IllegalArgumentException: setSelectedIndex: 1 out of bounds > at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622) > at net.sf.openrocket.gui.dialogs.ScaleDialog.init(ScaleDialog.java:358) > at net.sf.openrocket.gui.dialogs.ScaleDialog.<init>(ScaleDialog.java:235) > at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$3.actionPerformed(FreeformFinSetConfig.java:245) > at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) > at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) > at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) > at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) > at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) > at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) > at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) > at java.desktop/java.awt.Component.processEvent(Component.java:6400) > at java.desktop/java.awt.Container.processEvent(Container.java:2263) > at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) > at [... cut ...] d.java:113) > at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) > at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) > at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) 12520 524.527 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog 12521 524.527 INFO (SwingExceptionHandler.java:201) Showing Exception dialog 12524 0.000 WARN (-) ===== 134 WARN lines removed =====

jeffreyg-sh, 2022-07-07:
[image]

hcraigmiller, 2022-07-08:
The bug is more basic than stated. An attempt to scale any freeform fin causes an uncaught exception, even simply using a new design, adding a body tube, then adding a default freeform fin. Try to scale, uncaught exception.

However, you can scale the fin by itself using the Edit -> Scale function or the hover (over fin), right-click, scale function.

PR #1527 (2022-07-18): [#1520] Add check for selection options size
This PR fixes #1520.