openrocket/openrocket #1522: Feature request: Ability to add freeform fin sets to inner tubes (aerodynamically ignored)

hcraigmiller, 2022-07-09:
Quite a few users are using phantom body tubes for the attachment of shapes that are created using the freeform fin function. The most common use of this technique seems to be for the imitation of Estes style motor hooks. Where this technique uses two components, both types of which are intended to affect the aerodynamics of the model, a better method is needed.

The simplest method would seem to be the following:
1. Allow freeform fin sets to be attached to inner tubes.
2. When attached to an inner tube, a “Radial position” tab is added to that freeform fin set; this allows the freeform fin set to penetrate the outside plane of the inner tube (both ends of the motor hook).
3. When attached to an inner tube, that freeform fin set is ignored for aerodynamic purposes.
4. When attached to an inner tube, that freeform fin set displays a warning that it is being ignored for aerodynamic purposes (see warning on inner tube radial position tab).

This is what users are using the phantom body tube technique to visually create:

[image]

neilweinstock, 2022-08-14:
I am not enthusiastic about this change.  First of all, I think it is exactly one user that is doing this, not "users". 

It seems like it will be too easy for it to create confusion and/or lead users astray.

hcraigmiller, 2022-08-14:
Just as a point of clarrification, users refers to everyone who is using the K'Tesh motor mount files, and a number who create similar effects on their own.

As to confusion, what users are doing creates more confusion. . . see [TRF Discussion](https://www.rocketryforum.com/threads/misuse-of-phantom-body-tubes-in-openrocket-for-motor-mounts.174340/#post-2310242), center paragraph.

Just a suggestion. . .

neilweinstock, 2022-08-14:
I saw your thread, but I didn't see specifically what led you to post it.

I think I would feel better about this feature if, instead of using existing fin components, we created a new decorative object (I don't know what to call it, but it wouldn't be a "fin"), listed under "Internal Components".  Its configuration might function exactly like freeform fin if that's what makes the most sense.  This I believe would eliminate confusion.

neilweinstock, 2022-08-19:
I thought more about this and I am getting more enthusiastic about implementing it as a brand new internal component, called.... I dunno, maybe "Freeform Object"?

I'm not sure what it's parameters should be.  Obviously it would have a shape tab like freeform fins, and I would say radial positioning parameters like (e.g.) pods so it can float anywhere inside the airframe (and not be tied to a phantom body or something like that).  Oh, and a standard "position" parameter, and thickness like fins.

I'm not sure it needs the other fin parameters though, like number of fins and fin cant.  Since these are not fins.

Basically, it'll be a hybrid of some other components, tailored towards what it is: a randomly shaped object that can be placed anywhere inside the airframe.

neilweinstock, 2022-08-19:
"Freeform Mass Object", maybe?  Wondering if it should have calculated mass or just settable, like a standard mass object.

hcraigmiller, 2022-08-23:
I'm leaning towards a "Freeform Mass Component" being added to the _Mass Components_ component line, before the _Mass Component_ in the lineup. Compare suggestion of @neilweinstock, above.

hcraigmiller, 2022-09-16:
To revise my first post, in light of the comments of @neilweinstock, I suggest the following:

1. Create a new "Freeform Component" of the inner component type.
2. A Freeform Component may only be attached to inner tubes and coupler tubes. 
3. A Freeform Component configuration would include the General (field appropriate), Shape, Radial Position, Override, Appearance, and Comment tabs.
4. By including the “Radial position” tab, a freeform component may penetrate the outside plane of the inner tube (both ends of the motor hook).
5. A Freeform Component is ignored for aerodynamic purposes.
6. A Freeform Component does not trigger the "jagged edged fin" warning.

I think that covers everything.