openrocket/openrocket #1532: Feature request: Let 'Tab' keypress in component config dialog only focus text boxes, check boxes & combo boxes + have a more logical order

SiboVG, 2022-07-13:
Right now, when you select a text box in a component config dialog and press the 'Tab' key on your keyboard, the GUI switches focus to another element. However, the next focus element can be random + should IMO only apply to text boxes, check boxes and combo boxes, not for instance on sliders. This change will make it more easy to quickly switch editing parameters without having to click the GUI widget, or press tab a bunch of times until you're finally where you want to be.

So not like this:

https://user-images.githubusercontent.com/11031519/178818956-76a6caf8-4b57-42c6-b798-b011a5272482.mov

But something like this (I'm just clicking the boxes in this demo, but you get the idea):

https://user-images.githubusercontent.com/11031519/178819009-d6760c9d-3f24-42df-954f-8e3925e64e5c.mov

I think the most logical ordering would be to work column per column, so in the example of the body tube, first put focus on the text boxes for length, outer diameter, auto checkbox, inner diameter, wall thickness and finally filled checkbox, and then do the combo boxes on the right.

RemingtonHolder, 2022-08-14:
I might work on this if I find some extra time but im dropping this here for anyone that may work on this issue/feature: You have to modify the [focus traversal policy](https://docs.oracle.com/javase/7/docs/api/java/awt/FocusTraversalPolicy.html) in Swing

SiboVG, 2022-08-14:
> I might work on this if I find some extra time but im dropping this here for anyone that may work on this issue/feature: You have to modify the [focus traversal policy](https://docs.oracle.com/javase/7/docs/api/java/awt/FocusTraversalPolicy.html) in Swing

Well I've already started working on this a while back. Already wrote the focus traversal policy, only need to apply it to each config dialog.

SiboVG, 2022-08-14:
> Well I've already started working on this a while back. Already wrote the focus traversal policy, only need to apply it to each config dialog.

But I'll finish up the traversal policy for one config dialog and then you can get a crack on my branch at applying it to all other dialogs.

SiboVG, 2022-08-14:
@RemingtonHolder I created PR #1587. You can work directly on my branch and push your commits there. Just git clone my repo and switch to the issue-1532 branch (git command: `git clone git@github.com:SiboVG/openrocket.git && cd openrocket && git checkout issue-1532`).

So just take a look at how the traversal is implemented in the body tube config and apply the same to all other rocket components. The general tabs should already be covered (the override tab, shoulder tab, motor mount tab, appearance tab and comment tab), but best to check it twice.

RemingtonHolder, 2022-08-15:
Can you give me write access to your repo so I can push my commit, thanks!

SiboVG, 2022-08-15:
> @SiboVG Can you give me write access to your repo so I can push my commit, thanks!

Sent you an invite.

closed 2022-08-30 by SiboVG