openrocket/openrocket #1548: Support bilateral symmetry in instanced components (fins, pods, boosters)

neilweinstock, 2022-07-21:
Could have sworn I submitted a request for this in the past, but I don't see it anywhere so here it is.

Instanced components should have an option for bilateral symmetry, in which case:
1) The number of instances is 2
2) The "angle" parameter specifies offset of each instance from vertical
3) In the case of pods, the angle offset for child components is opposite on each side (whereas today they all go in the same direction)
4) from a CP standpoint, each set of two would need to be treated as two single instances, since there's likely no shortcut to creating an aggregate effect of the set.

I envision a checkbox for "bilateral symmetry" that, when checked, forces the number of instances to 2 and disables that field.

This would be a real time- and effort-saver for any airplane-like design (of which there are *many*).

hcraigmiller, 2022-07-22:
Would you describe/give examples of how OR functions now and how it would function with the suggested changed?

Though only one component bears the nomenclature, there are currently 7 components in 3 groups that have instances:
- Booster Set (Number of copies); Pod Set (Number of copies);
- Trapezoidal Fin Set (Number of fins); Elliptical Fin Set (Number of fins); Freeform Fin Set (Number of fins); Tube Fin Set (Number of fins); and
- Rail Button (Instance Count).

I am assuming that where you reference "airplane-like" designs that, of the foregoing components, Rail Buttons and Fin Sets are not within what you are calling "instances components." In other words, are you referring only to Booster and Pod sets when you say "instanced components"?

Is this an example of what you are trying to achieve, but with only 1 pod set, by using "bylateral symmetry"?

[image]

neilweinstock, 2022-07-22:
Rail buttons should be excluded, but fin sets are most definitely included. Consider if the wings in your image above have a dihedral or anhedral.  Consider all the fins on the Estes Interceptor.

In the present implementation, all those fins must be entered as two single fins, rather than as sets of two as would be more appropriate.

[Winglet Annotated]<https://user-images.githubusercontent.com/68821492/180341499-faa507ee-a381-4309-ae69-c0f379366133.png>

hcraigmiller, 2022-07-22:
So, is this closer to descibing what you are trying to achieve?

[image]

neilweinstock, 2022-07-22:
Exactly.

[Rotation Rev_01]<https://user-images.githubusercontent.com/68821492/180354164-56636d9a-00a0-4b01-8ace-4272170c4358.png>

neilweinstock, 2022-07-22:
I also made this image a full 5 years ago, the last time this was discussed:
[image]

neilweinstock, 2022-07-22:
Rear view of the Interceptor:
[image]

(courtesy of Chris Michielssen's blog)

neilweinstock, 2022-07-22:
Example UI mockup:
[image]

SiboVG, 2022-07-22:
The issue with re-using the 'angle' parameter for rotational offset between the two instances is that you cannot rotate the component around its parent anymore... I think the only possibility is to add an extra widget for the bilateral offset.

neilweinstock, 2022-07-22:
With bilateral symmetry I don't believe that rotation is necessary.

SiboVG, 2022-07-22:
> With bilateral symmetry I don't believe that rotation is necessary. There might be some very peculiar circumstances where it is useful, but they are (I believe) likely to be very rare and not worth additional UI complexity. The option proposed above is extremely *simple* and very useful in the vast majority of situations.

Probably true about the likeliness of needing a rotation feature when using bilateral instances, but I still wouldn't take away the possibility for it.

But my main issue with the UI proposition is that while the proposition is simple in the sense that you don't have to add an extra UI widget, to me it's not very intuitive that e.g. a 'Fin rotation' widget has the same functionality as 'Offset between bilateral instances'. I'm just a bit opposed to switching functionality of widgets, especially if the UI label does not at all represent the functionality...

neilweinstock, 2022-07-22:
While I understand what you're saying, _in my opinion_ it is clear enough in this case and anyone using it will understand very quickly.  To some degree, the angle of fins in bilateral symmetry could be thought of as fin rotation.  It's not really in conflict with the existing label, _in my opinion_.  The alternative is additional UI complexity.

If you really wanted then you could change the name of "Fin Rotation" to something like "Fin Angle" (?) when bilateral symmetry is enabled.

KTesh, 2022-07-22:
K'Tesh here...

I'm not quite up to speed on this issue, but here's my thoughts on how OR should function.  It should be able to replicate the designs we've seen coming out of the research departments of the manufacturers.  If a rocket is shaped like a plane, and has bilateral symmetry down the length of the rocket, or is more like a missile, and mirrored, OR should support it.  The minimum number of items a pod can support should be one...   If a rocket has a single fin off of the main body tube, and then other fins (or objects) jut off of it (such as in the case of rocket gliders), then so be it.  It should allow for whatever the designers have cooked up, and not limit us. 

Hope this helps.

JoePfeiffer, 2022-07-22:
Would "Mirrored" be a better term in the UI than "Bilateral Symmetry"? It might be clearer to the user that in this case defining a fin with a rotation of 45 degrees would result in one fin at 45 and one at -45.

neilweinstock, 2022-07-22:
Hmm, maybe.  Admittedly "Bilateral Symmetry" is a bit of a mouthful for a GUI, but it is descriptive and accurate.  Technically, bilateral symmetry here means precisely: "Mirrored around Y axis".  So if "Mirrored" is a shorthand description that is more palatable, I'd have no objection.

raketenolli, 2024-02-11:
Since the rocket would no longer be rotationally symmetric (3 or more axes of symmetry), we would have a range of CP as opposed to one fixed location according to Barrowman (or more elaborate forms thereof), depending on where the relative wind is coming from. Do we know how to handle that?

neilweinstock, 2024-02-11:
We already handle this situation, since you can have fins or pod sets with only 1 instance, creating a non-rotationally symmetric configuration.  So this proposed feature doesn't add anything new in that regard; it is merely a shortcut (albeit a *huge* one) when entering a bilaterally symmetric design.

raketenolli, 2024-02-11:
Ah, I just tried that with 23.09. It appears it only shows the most conservative CP position. I think that makes sense, even though my scientifically curious mind wants to see both locations of course.

neilweinstock, 2024-02-11:
Since this issue is being discussed again, I have two followups and one new item:
1) Terminology-wise, I don't care "mirrored" versus "bilateral symmetry".

2) Here is the current fin config:
[image]
For existing fin sets, "Fin rotation" means the angle off vertical of the first fin.  The remaining fins are space equally around the body.

On the other hand, for the proposed mirrored fin sets, "Fin rotation" would mean... the angle off vertical of the first fin.  The remaining fin is placed opposite to the first, mirrored around the vertical plane.

In other words, "Fin rotation" means exactly the same thing in both cases.  There is no reason I can see to have two separate parameters here.

3) A new thing that just occurred to me: as part of mirrored instances, the left/right decals should be flipped for the two instances, since that is the way 99.9999999% of designs would want it.

neilweinstock, 2024-02-11:
> Ah, I just tried that with 23.09. It appears it only shows the most conservative CP position. I think that makes sense, even though my scientifically curious mind wants to see both locations of course.

I agree, we could do better to show how CP varies with orientation.  I'm not sure how to do it cleanly in the GUI.  I seem to recall that Rocksim does do this?

JoePfeiffer, 2024-02-11:
We do show that in the component analysis.   What I'd like to do in the design window is show the current CP location as we rotate a side view.  There is concern that a user could rotate a design that's unstable in one axis but stable in another to make it look good...

neilweinstock, 2024-02-11:
My guess is that we should *always* show worst-case CP and stability margin in the upper right, but under the right conditions we could *also* show the varying value(s) as the design is rotated.

raketenolli, 2024-02-11:
Graphically, something like worst case in fully saturated red and current and/or best case in a lighter shade of red with a line connecting them (for the margin in the top-right corner always worst case)?

The component analysis CP view is not super intuitive because the set of fins that's determining the CP location at angle of attack 0° is the one you're looking at edge-on, but it's great to have that option at all. Those users that don't understand what's going on hopefully won't use it.

KTesh, 2024-02-11:
"A new thing that just occurred to me: as part of mirrored instances, the left/right decals should be flipped for the two instances, since that is the way 99.9999999% of bilaterally symmetric designs would want it."

By default the left/right decals should be flipped...  HOWEVER... In many cases, such as military jets with twin rudders, the outside of the rudders will *NOT* be mirrored (you'd end up with one side that has the text mirrored too).

KTesh, 2024-02-11:
Off topic, but related...  I'd love to see some kind of  reminder the user how to orient their decals in "Separate appearance for left and right side(s)" situations.  Something like a little icon, which may or may not pull up a pop up window, that shows the orientation of a theoretical fin, and what the text orientation should be.  Also, I'd love a pull down menu that would rotate paint and decal image (left or right) 90 degrees, or 180 degrees.  This would save a ton of editing when I mess up on my decals and paint

[image]