openrocket/openrocket #1554: Issue with motor configuration

zeke800, 2022-07-24:
Hello, 

Thank you so much for the great software!! I really enjoy using OpenRocket. 

I am using the latest beta version. When opening the motor dialog, it looks really weird and does not seem to load properly. 
[image]
How can I fix this?

Thank you!!

zeke800, 2022-07-24:
Here is the file by the way:
[rocket.zip](https://github.com/openrocket/openrocket/files/9176967/rocket.zip)

hcraigmiller, 2022-07-25:
Could not replicate.

[image]

> I am using the latest beta version. 

Which OS Platform, are you using the .jar or an installer, and where did you get it from?

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

zeke800, 2022-07-25:
Hello, I am using the Windows installer. I can confirm it works on the beta02 version, but not on the beta04 version. I got it from the official releases page. My system is Windows 10 Home.

[image]

hcraigmiller, 2022-07-27:
Have you tried to uninstall all of the beta versions, reboot, and reinstall beta 4?

zeke800, 2022-07-27:
Yes I have.

SiboVG, 2022-07-27:
Does this issue also persist in beta 3?

SiboVG, 2022-07-28:
Well this is interesting... I'm working on a fix for #1561 and while testing out some code, I was able to replicate this exact issue. So the issue has nothing to do with the setup of @zeke800.

[image]

SiboVG, 2022-07-28:
@zeke800 are you somewhat experienced with running OpenRocket from the JAR file? If so, could you verify whether [this JAR file](https://github.com/openrocket/openrocket/suites/7571532513/artifacts/313024235) fixes this issue?

PR #1565 (2022-07-30): [#1561] Such a stupid bug....
As the title dictates, very stupid bug. When implementing the window icon in #1447, I recycled some code from another window that has the window icon. So instead of using `GUIUtil.setDisposableDialogOptions`, `GUIUtil.setWindowIcons` should be used. This PR fixes #1561 and fixes #1554.
closed 2022-07-30 by SiboVG