openrocket/openrocket #1555: Don't automatically open first stage motor selection when duplicating motor configurations

neilweinstock, 2022-07-25:
See: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-4-is-now-available.171295/post-2302738

Right now OR automatically opens the first stage motor selection when creating a new motor configuration *or* duplicating an existing configuration.

Most likely this behavior is still appropriate and useful for new motor configs, but maybe not for existing ones.  I'm not sure if the best behavior would be to just don't open anything, or "take the hint" and open the selector depending on which cell was selected prior to the duplication.  My *guess* right now is that not opening anything automatically is the best approach here.

hcraigmiller, 2022-07-29:
I agree with @neilweinstock that this initially looks like an either or situation. 

1. Either **do not open** a sub-menu simultaneous with the _duplication_ of a configuration,
2. Or, **open** a sub-menu for the element that was in focus when the _Duplicate Configuration_ button was clicked.

One potential issue with option 2 is what action to take when the focus is on the "configuration" and not on an element of the configuration when the _Duplicate Configuration_ button is clicked, such as this:

[image]

When the focus is on the "configuration", there are, again, two options: either do not open a sub-menu at all, or open (my suggestion) the right-most element, when the _Duplicate Configuration_ button was clicked.

So, three choices:

1. **Do not open** a sub-menu simultaneous with the _duplication_ of a configuration.
2. **Do not open** a sub-menu simultaneous with the _duplication_ of a configuration **unless an element** of the "configuration" is in focus, then **open** the sub-menu for the same element that was in focus when the _Duplicate Configuration_ button was clicked.
3. **Open** the sub-menu for the same element that was in focus when the _Duplicate Configuration_ button was clicked, but, _if the "configuration" itself is in focus_, then **open the sub-menu for the right-most element** on the same tab as was in focus when the configuration was duplicated.

My preference would be Option 2 above.

neilweinstock, 2022-07-29:
I’ve thought about this a bit and favor option 1. This will satisfy user needs just fine. Anything else is over-thinking. IMHO.

[Configuration]<https://user-images.githubusercontent.com/68821492/181849191-b094dd06-39a1-4a50-83f0-7db15b262d60.png>

PR #1590 (2022-08-23): [#1555] Don't open config dialog for duplicate config
This PR fixes #1555.
closed 2022-08-23 by SiboVG