openrocket/openrocket #1569: Error

Matthewr7404, 2022-07-30:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.Finished making rocket went to hit new configuration and it gaave me this error
2.matthewprichamn@gmail.com
3.7372301232
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: setSelectedIndex: 0 out of bounds at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622) at net.sf.openrocket.gui.dialogs.motor.thrustcurve.ThrustCurveMotorSelectionPanel.updateData(ThrustCurveMotorSelectionPanel.java:477) at net.sf.openrocket.gui.dialogs.motor.thrustcurve.ThrustCurveMotorSelectionPanel.select(ThrustCurveMotorSelectionPanel.java:448) at net.sf.openrocket.gui.dialogs.motor.thrustcurve.ThrustCurveMotorSelectionPanel.selectMotorFromTable(ThrustCurveMotorSelectionPanel.java:604) at net.sf.openrocket.gui.dialogs.motor.MotorChooserDialog.open(MotorChooserDialog.java:123) at net.sf.openrocket.gui.main.flightconfigpanel.MotorConfigurationPanel.selectMotor(MotorConfigurationPanel.java:335) at net.sf.openrocket.gui.main.flightconfigpanel.FlightConfigurationPanel.newOrDuplicateConfigAction(FlightConfigurationPanel.java:146) at net.sf.openrocket.gui.main.flightconfigpanel.FlightConfigurationPanel.access$500(FlightConfigurationPanel.java:39) at net.sf.openrocket.gui.main.flightconfigpanel.FlightConfigurationPanel$NewConfigAction.actionPerformed(FlightConfigurationPanel.java:333) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.04
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\Matthew\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\AVRDUDE\avrdude-6.4-mingw32;C:\WinAVR-20100110\bin;C:\WinAVR-20100110\utils\bin;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\Program Files\Eclipse Foundation\jdk-8.0.302.8-hotspot\bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Gradle\gradle-7.2\bin;C:\WINDOWS\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\Users\Matthew\AppData\Local\Microsoft\WindowsApps;C:\Users\Matthew\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Matthew\AppData\Local\GitHubDesktop\bin;C:\Users\Matthew\Downloads\avrdude-v7.0-windows-x64\avrdude.exe;C:\AVRDUDE\avrdude-6.4-mingw32;C:\Users\Matthew\AppData\Local\Microsoft\WindowsApps;C:\Users\Matthew\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Matthew\AppData\Local\GitHubDesktop\bin;C:\Users\Matthew\Downloads\avrdude-v7.0-windows-x86\avrdude.exe;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.15+10-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.15
java.version.date=2022-04-19
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.15+10-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln12884425784343708990
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 11
os.version=10.0
[... 132 lines cut ...]
905402 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:InnerRadius]
905405 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:InnerRadius]
905408 3291.433 WARN (DoubleModel.java:951) Invalidating DoubleModel[InnerTube:Thickness] while still having listeners [javax.swing.JSlider$ModelListener@f2639df, javax.swing.plaf.basic.BasicSliderUI$Handler@27aafaf7]
905409 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:Thickness]
905412 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:Thickness]
[... cut ...]
905467 3291.433 WARN (DoubleModel.java:951) Invalidating DoubleModel[InnerTube:RadialPosition] while still having listeners [javax.swing.JSlider$ModelListener@1d7a81cc, javax.swing.plaf.basic.BasicSliderUI$Handler@4dc8fadc]
905468 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:RadialPosition]
905471 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:RadialPosition]
905474 3291.433 WARN (DoubleModel.java:951) Invalidating DoubleModel[InnerTube:RadialDirection] while still having listeners [javax.swing.JSlider$ModelListener@3e502cab, javax.swing.plaf.basic.BasicSliderUI$Handler@19a8ea7c]

Matthewr7404, 2022-07-30:
905475 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:RadialDirection]
905478 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[InnerTube:RadialDirection]
905479 3291.433 DEBUG (MemoryManagement.java:56) Adding object into collectable list: net.sf.openrocket.gui.configdialog.ComponentConfigDialog[dialog28,104,104,718x425,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title=Inner Tube configuration,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,702x386,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
905481 3300.439 DEBUG (BasicFrame.java:1810) Found frame net.sf.openrocket.gui.main.BasicFrame[frame0,-8,-8,2576x1456,invalid,layout=java.awt.BorderLayout,title=*Rocket (rocket shep.ork),resizable,maximized,defaultCloseOperation=DO_NOTHING_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,2560x1417,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true] for rocket Rocket
905502 3300.469 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905503 3300.469 USER (ThrustCurveMotorSelectionPanel.java:602) Selected table row 796: ThrustCurveMotorSet[Estes D12, type=Single-use, count=1]
905504 3300.469 USER (ThrustCurveMotorSelectionPanel.java:602) Selected table row 796: ThrustCurveMotorSet[Estes D12, type=Single-use, count=1]
905505 3300.469 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905506 3300.469 USER (ThrustCurveMotorSelectionPanel.java:602) Selected table row 796: ThrustCurveMotorSet[Estes D12, type=Single-use, count=1]
905507 3300.470 USER (ThrustCurveMotorSelectionPanel.java:602) Selected table row 796: ThrustCurveMotorSet[Estes D12, type=Single-use, count=1]
905508 3300.471 USER (ThrustCurveMotorSelectionPanel.java:602) Selected table row 796: ThrustCurveMotorSet[Estes D12, type=Single-use, count=1]
905509 3306.010 INFO (RocketPanel.java:874) using pre-existing simulation
905526 3306.038 DEBUG (GUIUtil.java:248) Clearing all models of window net.sf.openrocket.gui.dialogs.motor.MotorChooserDialog[dialog29,823,369,915x702,invalid,hidden,layout=java.awt.BorderLayout,APPLICATION_MODAL,title=Select a rocket motor,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,899x663,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
905527 3306.038 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905528 3306.038 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905529 3306.038 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905530 3306.038 USER (ThrustCurveMotorSelectionPanel.java:607) Selected table row -1, nothing selected
905533 3306.038 WARN (DoubleModel.java:951) Invalidating DoubleModel[MotorRowFilter:MinimumLength] while still having listeners [net.sf.openrocket.gui.dialogs.motor.thrustcurve.MotorFilterPanel$10@39c08e8b]
905534 3306.038 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[MotorRowFilter:MinimumLength]
905537 3306.038 WARN (DoubleModel.java:951) Invalidating DoubleModel[MotorRowFilter:MaximumLength] while still having listeners [net.sf.openrocket.gui.dialogs.motor.thrustcurve.MotorFilterPanel$11@b4328f]
905538 3306.038 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[MotorRowFilter:MaximumLength]
905539 3306.039 DEBUG (MemoryManagement.java:56) Adding object into collectable list: net.sf.openrocket.gui.dialogs.motor.MotorChooserDialog[dialog29,823,369,915x702,invalid,hidden,layout=java.awt.BorderLayout,APPLICATION_MODAL,title=Select a rocket motor,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,899x663,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
905540 3306.316 DEBUG (Simulation.java:371) Simulation: calling simulator
905541 3306.316 INFO (BasicEventSimulationEngine.java:88) >>Starting simulation of branch: Sustainer
905545 3306.316 INFO (BasicEventSimulationEngine.java:293) Queueing Ignition Event for: Inner Tube/21e4033f / D12 - Armed @: 0.0
905550 3306.316 INFO (BasicEventSimulationEngine.java:364) Igniting motor: Inner Tube/21e4033f / D12 - Armed @0.0
905554 3306.316 DEBUG (RK4SimulationStepper.java:190) Thrust estimate differs from correct value by 80.0%, estimate=1.3112049253731342 correct=0.2621428571428571 timestep=0.01, recomputing k1 parameters
906014 3306.327 DEBUG (BasicEventSimulationEngine.java:414) detected Motor Burnout for motor D12@ 1.65 on stage 1: Stage
906021 3306.327 INFO (BasicEventSimulationEngine.java:449) ==>> @ 1.65000; from Branch: Sustainer ---- Branching: Stage ----

906861 3306.342 DEBUG (FlightData.java:251) Computed flight values: maxAltitude=34.21813205467169 maxVelocity=18.429253208573687 maxAcceleration=50.3640360500909 maxMachNumber=0.0553323148966091 timeToApogee=3.449999999999995 flightTime=7.54999999999998 groundHitVelocity=10.391048915804987 laun [... cut ...]
909549 3382.715 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
909550 3382.715 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
909551 3382.715 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
909552 3385.087 USER (SwingExceptionHandler.java:220) User requested sending bug report
909553 3385.087 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

Matthewr7404, 2022-07-30:
[rocket shep (2).zip](https://github.com/openrocket/openrocket/files/9227591/rocket.shep.2.zip)

hcraigmiller, 2022-07-30:
It can be very difficult to replicate and diagnose issues without a description of what caused the error. Following is an example of how to describe what happened:

1. Open an example and go straight to the Flight simulations tab. 
2. Simulation 1 will be in focus, and left-click Run simulations to re-run that simulation (the simulation remains highlighted after being re-run and may be re-run any number of times without re-selecting the simulation).
3. Press Ctrl+A results in no selection; unexpected behavior, **the expected bahavior would be to select all simulations**.
4. Left-click on any simulation, including Simulation 1.
5. Press Ctrl+A results in selection of all simulations, this is the expected bahavior.

What did you do in your attached design file to cause the eror?

hcraigmiller, 2022-07-30:
The thrust of this issue (pun intended) appears to be that when a rocket is under-powered (not enough thrust to make it off the pad), the simulation hangs.

After removing the main body tube designation as a motor tube, adding a parachute, and selecting more powerful motors, the flight simulates (with under-powered motors, the simulation hangs).

[rocket shep.Rev_01.Under-Powered.zip](https://github.com/openrocket/openrocket/files/9227666/rocket.shep.Rev_01.Under-Powered.zip)
[rocket shep.Rev_02.zip](https://github.com/openrocket/openrocket/files/9227667/rocket.shep.Rev_02.zip)

JoePfeiffer, 2022-07-31:
I'm having a hard time understanding -- when I open the original .ork file I don't see any motors in either of the two configurations.  When I try to simulate it, I get (quite correctly) a popup telling me no motors are selected.  I'm not getting the crash being reported.

When I try the underpowered .ork, I do get the hang.  While that's a bug, I'm not sure it's the one the original user was trying to report.

hcraigmiller, 2022-07-31:
There were too many incorrect parts/settings to test the original file, and the user hasn't fully explained what the issue is. That was the best I could guess.

hcraigmiller, 2022-08-01:
The alternative issue, where the motors are being stripped upon opening the .ork file, is that the user is implementing custom motor configurations that are causing an error.

JoePfeiffer, 2022-08-15:
Don't fully understand what the actual issue was and haven't gotten clarification from the user.

closed 2022-08-15 by JoePfeiffer