openrocket/openrocket #1574: Photo Studio: no motor=no flame, add motor: no flame

jimmiedave, 2022-08-07:
Bring a model into Photo Studio without a motor. Turn on the Flame effect. You'll see no flame, although the non-existent flame will illuminate the rear of the rocket when turned on.

Without closing Photo Studio, select a motor for the rocket, and you'll still see the same effects (not) displayed.

Close Photo Studio and re-open it without changing anything and the flame appears (provided you had a motor and the flame effect selected before you closed it).

Tested Mac OS Catalina 10.15.7, OpenRocket v22.02.beta.04, Intel Mac: MacBook Pro 16in 2019, onboard Radeon Pro 5300M 4GB (and Intel UHD graphics 630 1536MB).

Tested first with my own X-15 model, and with A Simple Model Rocket.

**Flame Off**
[image]

**Flame On, showing illumination of the rocket by the invisible flame.**
[image]

hcraigmiller, 2022-08-10:
> Bring a model into Photo Studio without a motor. Turn on the Flame effect. You'll see no flame, although the non-existent flame will illuminate the rear of the rocket when turned on.

Behavior confirmed.

> Without closing Photo Studio, select a motor for the rocket, and you'll still see the same effects (not) displayed.

Behavior confirmed.

> Close Photo Studio and re-open it without changing anything and the flame appears (provided you had a motor and the flame effect selected before you closed it).

Behavior confirmed.

See also Issues #1122 (flame), #874 (smoke).

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

neilweinstock, 2022-08-13:
I believe this is one of many (!) known 3D breakages, all of which will have to wait until the whole 3D library is replaced.

SiboVG, 2022-08-14:
This issue is twofold:

- [ ] Disable flame lighting when no motors in the design
- [ ] Display motor in PhotoStudio when adding a new configuration with motor, while PhotoStudio was still active

jimmiedave, 2022-08-14:
The flame/no flame thing seems more to me like a switching issue: I'm guessing that there's some kind of method that's called (or some kind of event listener that's listening) whenever a parameter change is made, that refreshes photo studio. 
When trying to figure out what's going on, I took a known-good model and tried moving the motor mount way up in the body tube. The Photo Studio took the change immediately, changing the look of the flame. It probably needs to listen or switch when the motor is changed as well.

neilweinstock, 2022-08-14:
Sorry for my previous inaccurate comment. I was thinking of the smoke effects.

PR #1767 (2022-10-28): [#1574] Update PhotoPanel configuration on document change
This PR fixes #1574. Now when you add a new configuration while PhotoStudio is open, that configuration is drawn in PhotoStudio.

Steps to reproduce:
1. Open 'A simple model rocket'
2. Delete all the configurations
3. Open PhotoStudio
4. Add a new configuration and select a motor for it (PhotoStudio should still be open)
5. You can see the motor being rendered in PhotoStudio, which was previously not the case
closed 2022-10-28 by SiboVG