openrocket/openrocket #1576: Fin cant issues

sjh114961, 2022-08-09:
Need to be able to do a fin cant beyond 15 degrees. Many rocket gliders, rocket planes, and helicopter recovery model routinely use more than 15 degrees. Also, where is the pivot? I don't think it is on the leading edge.

hcraigmiller, 2022-08-10:
The fin cant pivot point is the center of the root chord. On a side note, I don't think I've ever seen a fin cant greater than 15 degrees; can you provide an example to satisfy my curiosity?

sjh114961, 2022-08-10:
Astron space plane. Astron Sky Dart, Gyroc
[GyrocImprovedRecovery.zip](https://github.com/openrocket/openrocket/files/9298827/GyrocImprovedRecovery.zip)

Rotating around root cord is very limiting. I would like, at least, to pivot at the leading edge.

hcraigmiller, 2022-08-10:
Is your suggestion that the fin cant pivot point be the intersection of the root chord and leading edge.

hcraigmiller, 2022-08-10:
> Astron space plane. Astron Sky Dart, Gyroc

Okay, I've looked at all of the designs that you provided, and, have a question. Would it be fair to say that the 15 degree cant is only being used after ascent and ejection, or, in other words, for recovery, not vertical flight?

sjh114961, 2022-08-11:
Yes, that would be true.

JoePfeiffer, 2022-08-12:
Letting the user specify the pivot point would seem to be a reasonable change.

Simulating helicopter recovery would be interesting.  I don't see it happening in this release cycle, though.