openrocket/openrocket #1579: Feature request: command line option(s) to suppress loading built-in databases

ehingsbe, 2022-08-10:
I have been an OpenRocket user since before v15.03, and find it very useful. However, I have had issues with the contents of the motor and parts databases. I would like to see a command-line (or configuration) option that allows the user to suppress automatic loading of those databases.

neilweinstock, 2022-08-13:
What kind of issues are you seeing, and how will not loading the databases fix your issues?

closed 2024-09-08 by SiboVG

ehingsbe, 2024-09-09:
I'd found inaccuracies in the databases (motor thrust curves that didn't
match those that the mfr supplied, and if I recall correctly, some parts DB
issues too).
Ultimately, of course, those errors should be fixed, but I don't have notes
on the specifics.
If the database tables were externally accessible, that could also be a
reasonable solution, but that seems like it could be disruptive.

Thanks!
Ed

PR #2555 (2024-09-08): [#1579] Add command-line argument to bypass preset and motor loading at startup
This PR fixes #1579. You can now bypass the component preset and motor database loading respectively with the command line arguments `-Dopenrocket.bypass.presets` and `-Dopenrocket.bypass.motors`. It has been documented on the GitHub Wiki: https://github.com/openrocket/openrocket/wiki/Command-line-arguments.