openrocket/openrocket #1581: Appearance texture scaling issue

acceleroto, 2022-08-12:
I'm using version 22.02.beta.04 on a Mac running macOS 12.4. When I add textures to rockets, the scaling of those textures isn't correct in on body tubes. It also appears texture scaling on fins is also sometimes incorrect. Nose cone textures appear to be scaled correctly. I have not tried other rocket component types (transitions, etc.).

To replicate the problem:
- Create a rocket or load an existing ork file.
- Open the Body Tube configuration window, select Appearance, deselect Use default, and select a Texture/From file.
- Change the rocket view type to 3D Finished. Observe the body tube textures incorrectly scaled.

IIRC in version 15, when I created a texture with pixel dimensions proportional to the circumference and length of a body tube, I could use X & Y scaling of 1.0 and the texture would be properly scaled to align with the body tube. I also don't recall this problem on beta 3, but I'm not certain of that.

The textures I have tried are all in png format. I have not tried other file formats.

I have attached an ork of a generic rocket along with several texture files and a screenshot. In this rocket, I assigned the body texture to the body tube and the scaling shows as incorrect (see screenshot). However, I assigned the fin texture to the nosecone and fins and those textures appear to be scaled correctly. If a problem with fin scaling isn't discovered when hunting for this problem, I can try to make a new rocket model that replicates the problem I am seeing.
[image]
[image]
[image]
[texture_test_rocket.ork.zip](https://github.com/openrocket/openrocket/files/9313624/texture_test_rocket.ork.zip)

hcraigmiller, 2022-08-12:
Tested using current unstable.
As to the body tube appearance, could not replicate issue.

1. Saved decal file, downloaded .ork file, deleted decal from .ork file, saved revised .ork file.
2. Open decal in art application and confirmed pixel count as 1540 horizontal by 9300 vertical, the same ratio as 5.06 (circumferance of 1.6 dia BT) by 30.37 inches (length of BT).
3. Opened revised .ork file, opened body tube appearance tab, unckecked "Use Default", and added decal to appearance.

Body tube appearance is as expected, both x/y Scale and x/y Offset.

[image]

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-08-12:
Was about to type "must be a mac thing", but I also can't replicate the issue on macOS 12.4
[image]

@acceleroto do you have an Intel Mac or an Apple Silicon Mac?

acceleroto, 2022-08-12:
Thanks for looking into this. I just installed beta 4 on a Windows 10 machine and a separate Mac & didn't see the problem on either of them. I thought it might be an OpenJDK problem, but I'm running the same OpenJDK 11.0.10 on both Macs.

I'm seeing this on an Intel Mac: Mac Pro 5,1; macOS 12.4; 2x 6-core Xeon x5690; RX 6600 XT GPU. I'll see dig into drivers/etc on this Mac to see if I can find what's driving the issue. I'll report back with what I find.

Screenshots from the 2nd Mac and Win10
[image]
[image]
:

SiboVG, 2022-08-12:
Thanks for the detailed report!

acceleroto, 2022-08-12:
Well, sorry for the wild goose chase.

acceleroto, 2022-08-13:
I spent the last 2 hours hunting this down with no luck. My Mac in question is an old Mac Pro (circa 2010) that I updated to macOS 12.4 using OpenCore Legacy Patcher and is also using a GPU that's 12 years newer than the machine. I can't fix it on this machine and I haven't been able to replicate it on any other machine. Other texture file formats don't fix it, system/driver updates don't fix it, rain dances don't fix it.

I haven't seen a graphics-related issue on this Mac before, but Frankenstein appears to be the problem, not OpenRocket.

It's my recommendation to close this issue...and kindly let me buy you all drinks the next time we meet. Sorry for the trouble.

neilweinstock, 2022-08-13:
Don't worry about it.  3D issues are often weird and inconsistent and hard to chase down.

Closing.

closed 2022-08-13 by neilweinstock