openrocket/openrocket #1582: Scaling free form fin causes exception

sjh114961, 2022-08-12:
When scaling freeform fin I get this. 

[image]

[Z6.zip](https://github.com/openrocket/openrocket/files/9328652/Z6.zip)

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: setSelectedIndex: 1 out of bounds at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622) at net.sf.openrocket.gui.dialogs.ScaleDialog.init(ScaleDialog.java:358) at net.sf.openrocket.gui.dialogs.ScaleDialog.<init>(ScaleDialog.java:235) at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$3.actionPerformed(FreeformFinSetConfig.java:245) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.04
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocketBeta4A\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocketBeta4A\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocketBeta4A\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocketbeta4a_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocketBeta4A\
install4j.exeDir=C:\Program Files\OpenRocketBeta4A\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocketBeta4A\.install4j\i4jruntime.jar;C:\Program Files\OpenRocketBeta4A\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocketbeta4a\jre
java.io.tmpdir=C:\Users\sj_h1\AppData\Local\Temp\
java.library.path=c:\program files\openrocketbeta4a\jre\bin;C:\Program Files\BellSoft\LibericaJDK-11\bin\;C:\Program Files (x86)\BellSoft\LibericaJDK-8\bin\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ANT\apache-ant-1.9.14-bin\bin;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Users\sj_h1\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\sj_h1\AppData\Local\Programs\Python\Python39\;C:\Users\sj_h1\AppData\Local\Microsoft\WindowsApps;;C:\Users\sj_h1\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\sj_h1\.dotnet\tools
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.15+10-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.15
java.version.date=2022-04-19
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.15+10-LTS
[... 285 lines cut ...]
7782 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
7783 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
7784 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
7785 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
7786 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
7787 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
7788 1352.771 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
7789 1352.771 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
7790 1352.771 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
7791 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
7792 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
7793 1352.771 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
7794 1352.771 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
7795 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
7796 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
[... cut ...]
7801 1352.771 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
7802 1352.771 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
7803 1352.771 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
7804 1362.181 USER (SwingExceptionHandler.java:220) User requested sending bug report
7805 1362.186 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

hcraigmiller, 2022-08-12:
Testing using Beta 4 replicates error, error only occurs on the Freeform fin shape tab when the scale button is clicked (all other scaling features function correctly.

Testing with current unstable functions as expected with no anomlous behaviors. **This appears to have been fixed in the current unstable.**

Recommend second verification before closure.

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-08-12:
Indeed, duplicate of #1520.

closed 2022-08-12 by SiboVG