openrocket/openrocket #1584: Fit not workink in 3D view

sjh114961, 2022-08-13:
Model is cut off in 3D view.
[image]

[Z6.zip](https://github.com/openrocket/openrocket/files/9331767/Z6.zip)

hcraigmiller, 2022-08-13:
This may be a settings issue. Have you followed these [after installation instructions](http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings)?

I do have a design question as well. Why are you using a phantom body tube for the centering rings and motor tube? Such a component misuse can prevent OR features from functioning correctly.

Could not replicate issue using Beta 4 or the current unstable.

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

sjh114961, 2022-08-13:
Yes, I did this, at least, in one of the prior betas. I did it again for this Beta and now it looks fine.

hcraigmiller, 2022-08-13:
If your complaint has been resolved, please close this issue.

sjh114961, 2022-08-13:
The whole motor mount is from K’Tesh. I just tacked it in. My guess is its necessary to add it to another model

hcraigmiller, 2022-08-13:
> The whole motor mount is from K’Tesh. I just tacked it in. My guess is its necessary to add it to another model.

I believe that K'Tesh likely has the best of the best when it comes to OpenRocket files. But, K'tesh uses a phantom body tube with a motor mount (centering rings and motor tube) **for presentation purposes** (it is placeholder for the internal components), _not for inclusion of that phantom tube into the model rocket itself_. 

When using a K'Tesh motor mount file, the inclusion of that motor mount in a new design is a two-step process: (1) copy the motor mount components (EXCLUDING the phantom body tube) from the K'Tesh design and paste the same onto the desired body tube in the new design; then (2) copy the phantom body tube with the engine hook from the K'Tesh design and paste the same on the new design (its placement is after the body tube onto which the internal motor mount components were pasted).

[image]

It appears that your issue has been resolved. **If so, please close this issue.**

closed 2022-08-14 by SiboVG