openrocket/openrocket #1585: Discussion Thread: Switch UI Systems

RemingtonHolder, 2022-08-14:
Ok, I'm going to preface this by saying that there is a lot on the plate for this project and especially for the next release and I get that but wanted to open this discussion thread because, after sifting through a number of Issues it seems to me that Java Swing is in part responsible or at the root of many problems. I have a proposition that this project switches to JavaFX or another UI system that would be more modern and would probably fix or, at least allow an easier fix for many of the UI issues that are popping up, especially with adding support for arm processors like Apple M1. Just a suggestion but one I feel would help the project in the long run.

SiboVG, 2022-08-15:
Can't believe I never looked into JavaFX more thoroughly before, it looks very promising. Also has some modern chart plotting libraries, **and** a 3D engine (haven't looked into the fine details as to how suitable it would be as a replacement for OR's 3D rendering, but still).

As of today, I see the following pros:
- Oracle has already declared that it will only keep Java Swing in maintenance mode. JavaFX is really the future of Java GUI, and has a lot of active development
- Would indeed solve some of the intricacies of Java Swing
- Good support for other libraries (e.g. graph plotting, 3D...)
- Their 3D library looks much more easy/high level than the intended 3D rewrite in Java LWJGL (but again, don't know if it can handle the somewhat complex mesh generation that OR needs for some rocket components).

Now for the con:
- It would be a lot of work to do a complete rewrite. There would first need to be a lot of testing: trying to rebuild some parts of OR to see if JavaFX would be a good replacement, e.g. remake the plotting mechanism, remake some basic other UI elements, do some 3D testing... And frankly, whilst I do believe JavaFX could be a good replacement for Java Swing, I'm afraid the current OR developer group isn't extensive enough to do the JavaFX rewrite. There are still a lot of other more important issues to be tackled. Combining that with a complete rewrite of the UI would probably delay frequent OpenRocket releases - something that we are trying to avoid.

There is also some discussion within the OR group of doing a complete UI overhaul. If that were to happen one day, I do think that would be the perfect opportunity to rewrite it in JavaFX.

So my conclusion: I do see OR running on JavaFX one day, given how modernized and actively developed it is, **BUT** if our developer group doesn't size up, I do not see a UI rewrite start in the next year or two.

There is however one factor that could speed up the integration of JavaFX. One of the biggest issues of OR is currently the 3D view. If the JavaFX 3D engine would prove to be as capable as Java LWJGL, but more high-level and developer-friendly, **then** we should get a crack at the UI rewrite.

RemingtonHolder, 2022-08-15:
Even though I am not a ardent contributor I completely agree with what you say, and I have not had any experience working with JavaFX 3d but just from working with the 2d aspect it seems to be a lot more user friendly than Swing, _albeit most all of my experience working in Swing lies with me contributing to this project_

SiboVG, 2025-03-23:
Switching to FlatLaf in the 24.12 release has actually fixed a lot of UI bugs, even while stilling running Swing. As such, I believe we can hold on to Java Swing for some time, so I'll close this issue.

closed 2025-03-23 by SiboVG