openrocket/openrocket #1596: Improvement: More user-friendly way of reporting bugs/feature requests

SiboVG, 2022-08-16:
As read [here on TRF](https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-4-is-now-available.171295/post-2311630), some users have trouble reporting bugs. We should try to improve the ease of bug-reporting (e.g. send the bug report to a mail or server somewhere, from which we can put it on GitHub, or find some way to automate GitHub issue creation).

I've also been wondering about the idea to add a 'survey' feature to OR. So that after e.g. the 30th time you open OpenRocket, you get a pop-up saying something like "Hey, do you like OpenRocket so far? Do you have suggestions to make it better"?

hcraigmiller, 2022-10-08:
Source Forge 2022-07-30:
Version 22.02 beta 04
Please let the bug report dialog info be copied so we can send an email.  The linkages force a use of mail programs thru windows which none of my students have or are allowed to use and we do not allow them access to github or other outside systems.

SiboVG, 2022-10-08:
> Source Forge 2022-07-30: Version 22.02 beta 04 Please let the bug report dialog info be copied so we can send an email. The linkages force a use of mail programs thru windows which none of my students have or are allowed to use and we do not allow them access to github or other outside systems.

I don't really understand, you can just select and copy the text from the bug report?

JoePfeiffer, 2022-10-09:
It's a little hard to read...  I think the requester is probably in a school environment, and the students aren't allowed access to email clients or off-campus web sites, so they want OR to include some rudimentary MUA that sends an email from within the program.  While firing off a bug report from within OR seems like a good idea, I don't really like a one-way bug reporting channel that we can't respond to and try to get more details on.