openrocket/openrocket #1599: Photo Studio scaling issue with High-DPI settings (Windows 10)

ReinhardR, 2022-08-18:
When High-DPI settings are enabled, the window scales with the DPI scaling settings, but the active drawing area doesn't.

The effect is proportional to the scaling factor. The screenshot was made with 150%.
[image]

The issues is not present in 15.03.

Windows 10 21H2, Intel HD driver 20.19.15.5126

hcraigmiller, 2022-08-18:
Have you made the post-installation settings adjustments (see [Wiki](http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings))?

ReinhardR, 2022-08-18:
I haven't done that before.
Enabling "Use Off-Screen Rendering" fixes the issue.

hcraigmiller, 2022-08-18:
That's great. . . please close this issue.

closed 2022-08-18 by ReinhardR