openrocket/openrocket #1611: Simulation fails to ignite motor

sjh114961, 2022-08-24:
Simulations fail and motor not detected.
[SST_ShuttleD.zip](https://github.com/openrocket/openrocket/files/9419004/SST_ShuttleD.zip)

JoePfeiffer, 2022-08-24:
In the case of the design triggering the bug, the booster stage is completely empty (should be noted the sustainer is actually named "Booster" and the booster is named "Glider")

hcraigmiller, 2022-08-24:
After playing with this design, OpenRocket appears to be functioning correctly with the exception that there is no warning that the aft-most stage does not have a motor.

Is this just a duplication/variation of Issue #1468.

JoePfeiffer, 2022-08-24:
It's a variation...  I think #1468 will turn out to be a special case of this one (need a warning if there's no motor in the bottom stage).  Shall we close that one and mark it as superceded by this?

hcraigmiller, 2022-08-24:
I would say that the two are sufficiently connected that fixing Issue #1611 also fixes Issue #1468, but not the other way around. I would close Issue #1468; this discussion is a sufficient reference to take a look there when resolving this issue.

PR #1832 (2022-11-18): When a simulation fails badly enough to be aborted, don't turn the exception into a warning
At present, when a simulation throws an exception it gets caught and turned into a warning.  Really, anything so bad the simulation can't finish ought to bring up the simulation error dialog.  This PR causes the "catch" code to record the abort in the flight event branch and then rethrow the exception.

Fixes #1611.  This is actually a more general solution than what I asked for there.

The attached .ork aborts because there are no motors configured to start at launch time.
[empty-booster.ork.txt](https [... cut]
PR #1860 (2022-12-05): Disallow stage separation at launch
Nobody has been able to think of a situation in which this is necessary or helpful, and it provides a way for new users to get in trouble.

Fixes #1611
closed 2022-11-18 by SiboVG