openrocket/openrocket #1613: [Bug] Transition fin tab auto calculate errors

hcraigmiller, 2022-08-25:
If the transition ends are the same, then the fin tab auto calculate works correctly. But if either end is smaller than the other, the auto calculate does not function correctly.

[image]

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

hcraigmiller, 2022-08-25:
The fin tab function also results in the root edge being drawn incorrectly on curved transitions.

[image]
.

SiboVG, 2022-08-25:
An additional bug that I now notice (already present in beta 4) is that the auto calc functionality doesn't work when the centering rings are a child of the inner tube instead of a direct child of the fin's parent tube.

hcraigmiller, 2022-08-25:
> An additional bug that I now notice (already present in beta 4) is that the auto calc functionality doesn't work when the centering rings are a child of the inner tube instead of a direct child of the fin's parent tube.

I'm not sure that doing so would have the best outcome. The way it is, the user can nest one motor mount inside of another and the fin tab uses the direct child of the fin's parent tube approach to identify the correct centering rings. Wouldn't allowing the identification of centering rings that are a child of the inner tube defeat this ability?

SiboVG, 2022-08-25:
> > An additional bug that I now notice (already present in beta 4) is that the auto calc functionality doesn't work when the centering rings are a child of the inner tube instead of a direct child of the fin's parent tube.
> 
> I'm not sure that doing so would have the best outcome. The way it is, the user can nest one motor mount inside of another motor mount (inner tube and centering rings) and the fin tab uses the direct child of the fin's parent tube approach to identify the correct centering rings. Wouldn't allowing the identification of centering rings that are a child of the inner tube defeat this ability?

Hm, you may be right, didn't think about it that way.

PR #1614 (2022-08-26): [#1613] Fix bugs in fin tabs auto calc
This PR fixes #1613. The following bugs are fixed:
- tab auto calc works for non-equal sized centering rings
- only use direct children of the fin's parent for the auto-calculation
- there was an issue where if a transition's aft or fore diameter differed, that the fin's height wasn't calculated properly

I did replicate @hcraigmiller's issue of the root edge not being drawn correctly.
closed 2022-08-26 by SiboVG