openrocket/openrocket #1618: [Bug] When fin tabs are used on curved nose cones and transitions, the fin root edge is not being drawn correctly

hcraigmiller, 2022-08-26:
[image]

hcraigmiller, 2022-08-26:
Nose cone behaves similarly. . .

[image]

[Fin Tab Errors.txt](https://github.com/openrocket/openrocket/files/9434601/Fin.Tab.Errors.txt)

SiboVG, 2022-08-26:
Current unstable branch. Are you sure you're using the latest unstable?
[image]

This issue should have been resolved by PR #1598.

hcraigmiller, 2022-08-26:
The issue is not with the fin tab itself (the fin tabs function correctly), it is that if a fin tab is used on a curved nosecone or transition then the root chord that is not part of the fin tab does not follow the curve (it is straight). I shorthened the fin tab length to make the error more pronounced.

SiboVG, 2022-08-26:
Ah now I see, it happens if the tab height is non zero and the tab length is smaller than the fin length.

PR #1620 (2022-08-28): [#1618 & #1597] Correct fin tab display on curved parent tube & fixes 3D issues for fin tabs
This PR fixes #1618 and fixes #1597.
closed 2022-08-28 by wolsen