openrocket/openrocket #1626: Support beveled fin profiles

neilweinstock, 2022-08-30:
HPR rockets commonly use fins beveled to a sharp edge.  It would be good to support this fin profile natively inside OR, rather than force the user to try to figure out which of the available profiles will behave most similarly to the bevels.

Profiles such as Nike fins (diamond shaped profile) might or might not behave similarly enough to front and rear bevels that one option will cover all of it.  Any explicit acknowledgement of beveled profiles would be an improvement.

hcraigmiller, 2022-08-30:
Background -- RASAero II features the following fin airfoils:

- Hexagonal Airfoil (leading and trailing edges)
- NACA Airfoil (full airfoil)
- Double-Wedge Airfoil
- Biconvex Airfoil
-

neilweinstock, 2023-02-23:
I wonder if it would be good enough here to simply say "airfoiled/beveled" and leave it at that?  From a drag perspective they may not be different enough to worry about.