openrocket/openrocket #1629: Show calculated Mass, CG, and CD values for non-overridden components

JoePfeiffer, 2022-08-30:
In the Override tab, if a component isn't overridden then an override value of 0 is shown for all three values unless changed by the user. Showing the calculated values would give the user a starting point for overrides.

SiboVG, 2022-09-01:
Should the CD be taken at Mach 0.30? (since the component analysis CD is also at Mach 0.30 by default).

JoePfeiffer, 2022-09-01:
All the various displayed aerodynamics' Mach numbers should be coordinated -- I *think* the right place to set this is in the component analysis window (the only two places I can remember to change it are in the preferences, which sets the default, and in the component analysis window).  Ah, just noticed changing it in the component analysis window doesn't change it in the rocket view (but changing the wind direction and angle of attack does).

JoePfeiffer, 2022-09-01:
Correction -- it changes the calculation, it just doesn't change the displayed mach number

PR #2237 (2023-07-05): [#1629] Show calculated mass, CG & CD in override tab if not overridden
This PR fixes #1629. When the mass, CG or CD of a component is not overridden, the override values will be the calculated values.

Demo:

https://github.com/openrocket/openrocket/assets/11031519/4f1ead16-5b26-4c78-96e0-189f8b773177

Note: when toggling the override CD the simulation results can slightly change. This is because in the overridden state, the individual CD components (pressure, base & friction CD) are set to 0 and only the total CD is decisive, so I think that's the reason for the very small difference.

---

 [... cut]
closed 2023-07-05 by JoePfeiffer