openrocket/openrocket #1631: Icons to show components in tree overridden by ancestor

JoePfeiffer, 2022-08-31:
At present the component tree shows when the CG or mass of a component is directly overridden.  It ought to also show when it's being overridden by an ancestor (perhaps using the same icon, but in a light blue?) and when its override is set, but it's being ignore because it's being overridden by an ancestor (perhaps using the same icon, but in grey?).

It'll be enough easier to implement this after implementing #1628 that it should wait until that's been done (or a decision has been made not to do it).

SiboVG, 2022-09-02:
Would it be sane to also disable the child override checkbox, and add an info text e.g. "Mass and CG overridden by parent"? The info text could be implemented as a hover text over the checkbox, but I think an extra info label above all the checkboxes would be best.

JoePfeiffer, 2022-09-02:
> Would it be sane to also disable the child override checkbox, and add an info text e.g. "Mass and CG overridden by parent"? The info text could be implemented as a hover text over the checkbox, but I think an extra info label above all the checkboxes would be best.

The semantics of component overrides strike me as really screwy, but seem to work.  Checking the "override subcomponents" box overrides the subcomponent values for any parameter that's overridden, but not those that aren't.  So (thinking of the simple model rocket here) it's entirely possible to override CD for the stage, CG for the booster, and mass for a finset, while overriding subcomponents for each of the three.  So we don't want to disable subcomponent overrides for the subcomponents.

If we ever start having users clamoring to be able to override the body tube's mass and CG, and override subcomponents mass but not CG, we'll have to think about individual checkboxes for subcomponent overrides of each parameter.  In that case, we may want to rethink disabling subcomponent overrides for overriden subcomponents.

hcraigmiller, 2022-09-22:
Wouldn't the most recognizable new icon simply be the old icon with a red circle and slash over it?

JoePfeiffer, 2022-09-22:
> Wouldn't the most recognizable new icon simply be the old icon with a red circle and slash over it?

My guess is using saturation to indicate it would be equally recognizable (though I'm torn on which should be light blue and which should be grey!).  I'm with @SiboVG that the circle/slash would look like it was indicating a user error or something.  Actually, even though it makes accessibility problematic, I like trying to reserve red for some sort of serious error.

closed 2022-11-14 by JoePfeiffer