openrocket/openrocket #1632: [Usability]  Selecting configuration vs. motor is confusing in simple case

broofa, 2022-08-31:
**Problem:**

When working with single-motor rockets - which I assume is the majority use case by far - making a selection in the motors & configuration tab is actually rather confusing.  For example, the difference between this:
[image]

... versus this:
[image]

... is subtle.  The two cells are labeled identically.  They both appear to represent the same motor.  Yet only one of these disables the motor buttons.

**Suggested solution:**

When a rocket has a single motor there should be no difference in how these selections behave.  Both should implicitly select both the configuration and the motor.  I.e. The motor buttons should be enabled in both cases.

SiboVG, 2022-08-31:
> When a rocket has a single motor there should be no difference in how these selections behave. Both should implicitly select both the configuration and the motor. I.e. The motor buttons should be enabled in both cases.

Hm, I'm not sure if this is how it should be. Could be confusing for users that in some scenarios (single-motor designs), the configuration column acts differently than on multi-motor designs. But I agree that this behavior can be confusing. Maybe it would even be better to remove the configuration column from that table entirely, and move it somewhere else, in a separate table. I guess that's something to think about when we're doing a complete UI overhaul of OR.

hcraigmiller, 2022-08-31:
The better and more consistent approach would be to block the configuration column from being selected. This will require that a "New Configuration" option be added to the MMT column selected motor drop down in order to replace the functionality lost by blocking the configuration colmn.

SiboVG, 2022-08-31:
> The better and more consistent approach would be to block the configuration column from being selected. This will require that a "New Configuration" option be added to the MMT column selected motor drop down in order to replace the functionality lost by blocking the configuration colmn.

I'm also not too keen on this approach. If you want to rename, delete, or duplicate a configuration, I find it more intuitive to select that configuration in the config column. Blocking selection in that column would block that habit.

hcraigmiller, 2022-08-31:
> I'm also not too keen on this approach. If you want to rename, delete, or duplicate a configuration, I find it more intuitive to select that configuration in the config column. Blocking selection in that column would block that habit.

My initial reaction was that this is a training issue, not a coding issue, and I didn't see your post until after mine was posted. My suggestion was based on the current MMT column selected motor drop down already having options for rename, "remove", and duplicate. I'm still not convinced that this is an issue beyond training.