openrocket/openrocket #1633: [Usability] Numeric inputs should accept values with explicit units

broofa, 2022-08-31:
It would be great if the numeric input fields throughout the app allowed the user to enter values with explicit units, and converted automatically.  For example, if my default length unit is inches, I'd like to be able to type "10 cm" into a dimension field and have it automatically converted to "3.94".

Editorial: There's no consistent units of measure used in rocketry. At least, not in the US.  It's all well and good to have configurable preferences, but when entering values the units I use depend on a bunch of factors.  What system did the manufacturer use?  Who am I discussing the design with? What measuring device do I happen to have on hand ... and so on.

For example, my preference is to use metric units throughout, but most of my tape measures are in feet and inches.

JoePfeiffer, 2022-09-06:
I'm not sure how OR's current input fields don't support your needs.  I just checked, and yes: with units set to Imperial, I can add a nose cone; in the component configuration dialog I can set the units of Length to cm using the dropdown and enter a value. It's true the next time I open the configuration window it's reported in inches, but that doesn't seem unreasonable.
[image]

broofa, 2022-09-06:
@JoePfeiffer Thanks for the quick response.

I wasn't actually aware you could click the unit label to change it.  I guess you could chalk this up to a "training" issue (and feel free to close this is that's what you decide)... but I think this is still a UI problem, for two reasons ...

First, it never occurred to me to click that unit label because, simply put, it doesn't look clickable.  It just appears to be nothing more than a label showing me what units the value is in.  It would be less jarring if that was the look of drop-downs throughout the app, but even w/in the same panel other drop-downs are much more obviously clickable: 
[image]

Second, a unit drop-down isn't easy to use.  When entering a numeric value you're using the keyboard, but changing the units requires moving your hand to the mouse.  (The unit label isn't even tab-navigable, so there's no way to change it w/out using the mouse.)  This makes it rather awkward to change, especially if you're dealing with multiple fields.

**Suggestion**
IMHO, the current numeric control is too complicated and does not function well.  There are 4 (**four!**) separate controls for each field, each with their own peculiarities. Specifically:

[image]

1. Numeric text field: doesn't support typing in values with units.  (Nor does it warn when entering invalid value except by clearing the value you entered once focus changes.  This forces the user to re-type the value.  😢 )
2. Increment/decrement buttons: Not tab navigable, only increments by +/-1, regardless of whether this is appropriate for the field value.  (E.g. If the field value is `0.015`, adding/subtracting one is _never_ going to be what the user wants.)
3. Unit field: Not tab navigable, not obviously clickable.
4. Slider control: Tab navigable (yay!), but doesn't respond to keyboard input in any meaningful way... which begs the question, "Why can we tab-navigate to this?!?"

_All of which is to say, I think a simple text field with a bit of "smarts" to it would be a much more effective interface.  For example:

[image]

* Accept values with explicit units, as mentioned previously, values are converted to default units on ENTER, TAB, or focus change.
* ... without units, values use the default units (always shown in label)
* Up and down arrows increment based on value precision.  E.g. "0.12" will change by +/- 0.01, "1.0" changes by +/-0.1.  "123" changes by 1... etc.

Such control would be simpler to implement (don't need to sync state across varied controls), simpler to use (no need to move hands away from keyboard, less complex tab-navigation chain), takes up less space, and involves less "visual clutter" in the UI.

That's my $0.02.  Thanks for listening. :-)

JoePfeiffer, 2022-09-07:
Ah,  now I see what you're getting at.  I'm leery of "smart" parsing of units; I've seen that turn into a mess. All the same, the units dropdown ought to be more evident, and it ought to be possible to tab to it.  Now that you mention it, I'm not sure I've ever used the sliders, and there's no good reason tab navigation to go to them in any case...

SiboVG, 2022-09-07:
> 4\. "Why can we tab-navigate to this?!?"

Is already disabled in the latest unstable branch. Now you can only tab **often used widgets** such as text inputs, buttons, and dropdown menus.

> it ought to be possible to tab to it

I disagree, as it rarely happens that users bother to change the units compared to how many times they want to simply parse text boxed by tabbing. Adding the unit box to the tab traversal policy would only add an annoyance for users who want to quickly parse through the config dialog.

hcraigmiller, 2022-09-08:
I'm not in favor of the suggested change to the interface. At some level there has to be a balance between what the most common use of the interface is and making changes that very few users would actually benefit from, especially when the change adds no new functionality, it just makes the interface slightly "smarter" at the risk of creating more confusion (what was the input abbreviation for gram, "g" or "gm").