openrocket/openrocket #1640: [Feature request] Add radial position to launch lugs and rail buttons

hcraigmiller, 2022-09-03:
In real life, launch lugs and rail buttons are frequently attached to standoffs. Currently, the only way to replicate this in OpenRocket is by using a pod, which requires the use of a phantom body tube. Adding radial positioning to these two components fixes that problem by allowing the launch guide to be positioned on the standoff without the use of a pod.

JoePfeiffer, 2022-09-03:
Would it be reasonable to also have a checkbox that would put a rectangular standoff between the launch lug and the airframe?

hcraigmiller, 2022-09-03:
> Would it be reasonable to also have a checkbox that would put a rectangular standoff between the launch lug and the airframe? User interface would present a checkbox for a standoff and a spinner for a standoff height?

If a "Shape" tab could be added to the launch lug and rail button component configurations so that the user could create a freeform standoff between the rail guide and the body tube, you would have the best of all worlds.

JoePfeiffer, 2022-09-03:
Freeform makes it harder... but width (default same as lug), length (default matches lug) and height would cover just about every standoff I've ever seen.  I've seen an occasional cylindrical standoff, but just rectangular would be a good start.

I think the exisitng base height parameter on rail buttons covers all the buttons I've ever seen (specifically buttons, not all the other variations on rail guides out there).

hcraigmiller, 2022-09-03:
> Freeform makes it harder... but width (default same as lug), length (default matches lug) and height would cover just about every standoff I've ever seen.

The reason radial positioning or freeform standoffs are important is for designs like that in Issue #1639; the launch lug is on top of the tip chord of a freeform fin.  Without radial positioning or freeform standoffs, the user is stuck with using a pod and phantom body tube to achieve the desired design.

As to the companies producing angled front faces on standoffs (such as Binder Design, below), a flat surface would likely not simulate much differently.

[image]

JoePfeiffer, 2022-09-04:
Ah, now I see the issue.... radial positioning is probably the most flexible....