openrocket/openrocket #1646: Fin appearance goes transparent when fin tabs used

sjh114961, 2022-09-06:


sjh114961, 2022-09-06:
Using these 2 models as reference, when the fin tabs are used the fins go transparent even if opacity set to 100%. Both models in same zip file the R1 version has no tabs the other model does.
[v2_18mm.zip](https://github.com/openrocket/openrocket/files/9500514/v2_18mm.zip)

SiboVG, 2022-09-06:
The issue you're talking about is a duplicate of #1597, which is already fixed and will be included in beta 5.

**However**, when testing this on the latest unstable branch, I noticed that the fins go "transparent" when the transition is non-conical, even when they have no tabs. So there is a new issue created due to some of the recent bug fixes in fins on non-conical transitions.

sjh114961, 2022-09-06:
One of the models in the zip file does not have the fin tabs and is NOT conical and seems to work.

SiboVG, 2022-09-06:
> One of the models in the zip file does not have the fin tabs and is NOT conical and seems to work.

That's for beta 4, not for the latest unstable branch of this repo.

hcraigmiller, 2022-09-07:
If you scale the rocket 200% (fin tabs and no fin tabs), this is what happens with the parachute _**and fins**_:

[image]
[image]

As to the parachute, current unstable does the same thing, I'll create a new parachute issue?

Build 992

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

PR #1647 (2022-09-10): [#1646] Increase fin root resolution
This PR fixes #1646. Turns out the fin root points resolution of 2.5 mm was too small. Increasing it to 5 mm appears to have solved the issue.

So fin template export will be **slightly** less smooth (see discussion [here](https://github.com/openrocket/openrocket/pull/1598#issuecomment-1218776893)).
closed 2022-09-10 by SiboVG