openrocket/openrocket #1661: [Bug] Rail buttons do not scale (size or instance separation)

hcraigmiller, 2022-09-11:
[image]

[Simple.RB.Rev_01.ork.txt](https://github.com/openrocket/openrocket/files/9541596/Simple.RB.Rev_01.ork.txt)

JoePfeiffer, 2022-09-11:
I don't know how one would reasonably scale the size of rail buttons.  They stay the same size as you scale the rocket until you decide you really need to go up a size, or can afford to go down a size.

Separation should, however, scale, especially since lots of people anchor them to bulkheads and centering rings.

hcraigmiller, 2022-09-11:
I completely undestand @JoePfeiffer's position. However, a couple of points: (1) not scaling the size of rail buttons would treat rail buttons different than any other component existing in OpenRocket, including launch lugs which one could agrue should not be scaled for the same reason expressed concerning rail buttons; and (2) when a rocket is scaled, nearly all, if not all, of the components need to be modified to exising available parts, and the scaled part dimensions can be used as the starting point to identify the correct replacement.

I believe that the best approach is to scale the rail buttones so that the scaled dimensions can be similarly be used to identify the most appropriate replacement rail button size, even thought there are only four sizes to chose from before you self-manufacture. As I indicated, launch lugs are in that same boat, with 1/8", 3/16", 1/4" and 1/2" diameter being the standard sizes, but launch lugs are scaled, as I believe rail buttons should be.

SiboVG, 2022-09-11:
I get Joe's point of view, but agree with Craig that there should at least be the possibility to scale rail buttons. Users can always multi-select every component except for the rail button if they want to scale every component except for the rail button. (funnily enough, while writing this, I found another bug... #1663)

JoePfeiffer, 2022-09-11:
Good point.  Long term, I think we need to evaluate what should and what should not be scaled -- for instance, body tube thickness, fin thickness, launch lug diameter should not.  But that's a conversation for another day.  For the moment, you're right, we should go for consistency.

PR #1664 (2022-09-12): [#1661] Add scaling for rail buttons
This PR fixes #1661 and adds scaling for rail buttons.
closed 2022-09-12 by SiboVG