openrocket/openrocket #1667: Open Rocket GUI Glitches After Accessing Menu

Phyronian, 2022-09-11:
I've been encountering a visual glitch after selecting a nose cone and clicking on the nose cone shape menu.  Before clicking the drop-down menu, the program appears normal, but after clicking the menu this happens:
[image]
[image]

I have tried uninstalling the latest beta version and installing other versions, installing a local java version, and then installing the versions, making sure my windows is completely up to date, but nothing works.  I'm unsure what to try at this point.

hcraigmiller, 2022-09-11:
This may be a known local settings issue. Have you followed these instructions:

- [Wiki -- Settings Check](http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings)
- [Wiki -- Graphics Settings](http://wiki.openrocket.info/FAQ#Windows_Graphics_Settings)

Phyronian, 2022-09-11:
I tried the "settings check" actions, but did not fix the issue.
The second link, "Graphics settings" seems to be very similar to my issue in appearance, however, when designating openrocket.exe to use integrated graphics only, the issue still persisted.

hcraigmiller, 2022-09-11:
Please post the bug report, thanks for running the checks.

Phyronian, 2022-09-12:
There's no bug report generated by the software, is there a way to force it to generate one?

hcraigmiller, 2022-09-12:
Start OpenRocket, trigger the error in as few steps as possible, then left-click Help > Bug report.

Phyronian, 2022-09-13:
---------- System information ----------
OpenRocket version: 22.02.beta.04
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\Phyronian\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Eclipse Adoptium\jdk-17.0.2.8-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\dotnet\;C:\Users\Phyronian\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Phyronian\AppData\Local\Programs\Python\Python39\;C:\Users\Phyronian\AppData\Local\Microsoft\WindowsApps;;C:\Users\Phyronian\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Phyronian\.dotnet\tools
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.15+10-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.15
java.version.date=2022-04-19
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.15+10-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln1865966025323906748
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=c:\program files\openrocket\jre\bin
sun.cpu.endian=little
sun.cpu.isalist=amd64
sun.desktop=windows
sun.io.unicode.encoding=UnicodeLittle
sun.java.command=C:\Program Files\OpenRocket\OpenRocket.exe
sun.jnu.encoding=Cp1252
sun.management.compiler=HotSpot 64-Bit Tiered Compilers
sun.os.patch.level=
user.country=US
user.dir=C:\Program Files\OpenRocket
user.home=C:\Users\Phyronian
user.language=en
[... 121 lines cut ...]
435 11.112 VBOSE (DoubleModel.java:947) Invalidating DoubleModel[NoseCone:AftShoulderRadius]
436 11.112 WARN (DoubleModel.java:951) Invalidating DoubleModel[NoseCone:AftShoulderRadius] while still having listeners [javax.swing.JSlider$ModelListener@5d4191ab, javax.swing.plaf.basic.BasicSliderUI$Handler@59b3f0f1]
437 11.112 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[NoseCone:AftShoulderRadius]
438 11.112 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderRadius] removing listener (total 0): javax.swing.JSlider$ModelListener@5d4191ab
439 11.112 VBOSE (DoubleModel.java:947) Invalidating DoubleModel[NoseCone:AftShoulderRadius]
440 11.112 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[NoseCone:AftShoulderRadius]
441 11.112 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderLength] removing listener (total 2): javax.swing.JSpinner$ModelListener@2656d343
442 11.112 VBOSE (DoubleModel.java:947) Invalidating DoubleModel[NoseCone:AftShoulderLength]
443 11.112 WARN (DoubleModel.java:951) Invalidating DoubleModel[NoseCone:AftShoulderLength] while still having listeners [javax.swing.JSlider$ModelListener@4b8b772b, javax.swing.plaf.basic.BasicSliderUI$Handler@1637514f]
444 11.112 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[NoseCone:AftShoulderLength]
445 11.112 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderLength] removing listener (total 0): javax.swing.JSlider$ModelListener@4b8b772b
446 11.112 VBOSE (DoubleModel.java:947) Invalidating DoubleModel[NoseCone:AftShoulderLength]
447 11.112 DEBUG (MemoryManagement.java:56) Adding object into collectable list: DoubleModel[NoseCone:AftShoulderLength]
448 11.112 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderThickness] removing listener (total 2): javax.swing.JSpinner$ModelListener@2954e553
449 11.112 VBOSE (DoubleModel.java:947) Invalidating DoubleModel[NoseCone:AftShoulderThickness]
[... cut ...]
456 11.113 VBOSE (Rocket.java:446) Removed ComponentChangeListener net.sf.openrocket.gui.configdialog.ComponentConfigDialog[dialog0,26,26,625x452,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title=Nose Cone configuration,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,609x413,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true], current number of listeners is 41
457 14.206 USER (BasicFrame.java:882) Bug report selected
458 14.208 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

Skyfinder84, 2022-09-21:
SAME ISSUE TO ME:
Include detailed steps on how to trigger the bug: 
(You can edit text directly in this window) 
1. Open Software
2. Select Nose
3. Select Shape
4. GUI glitch, multiple clone copy of screen’s segment.
What does the software do and what in your opinion should it do in the case described above: 
Include your email address (optional; it helps if we can contact you in case we need additional information): 
(Do not modify anything below this line.) 
---------- System information ---------- 
OpenRocket version: 22.02.beta.04 
OpenRocket source: default 
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar 
JOGL version: 2.4.0-rc-20210111 
Current default locale: it_IT 
System properties: 
awt.toolkit=sun.awt.windows.WToolkit 
exe4j.consoleCodepage=cp0 
exe4j.isInstall4j=true 
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe 
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe 
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0 
exe4j.tempDir= 
exe4j.unextractedPosition=0 
file.encoding=Cp1252 
file.separator=\ 
install4j.appDir=C:\Program Files\OpenRocket\ 
install4j.exeDir=C:\Program Files\OpenRocket\ 
install4j.launcherId=59 
install4j.swt=false 
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment 
java.awt.printerjob=sun.awt.windows.WPrinterJob 
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar 
java.class.version=55.0 
java.home=c:\program files\openrocket\jre 
java.io.tmpdir=C:\Users\Niro_\AppData\Local\Temp\ 
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files\Oculus\Support\oculus-runtime;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\Calibre2\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Users\Niro_\AppData\Local\Microsoft\WindowsApps 
java.runtime.name=OpenJDK Runtime Environment 
java.runtime.version=11.0.15+10-LTS 
java.specification.name=Java Platform API Specification 
java.specification.vendor=Oracle Corporation 
java.specification.version=11 
java.util.Arrays.useLegacyMergeSort=true 
java.vendor=BellSoft 
java.vendor.url=https://bell-sw.com/ 
java.vendor.url.bug=https://bell-sw.com/support 
java.version=11.0.15 
java.version.date=2022-04-19 
java.vm.compressedOopsMode=Zero based 
java.vm.info=mixed mode 
java.vm.name=OpenJDK 64-Bit Server VM 
java.vm.specification.name=Java Virtual Machine Specification 
java.vm.specification.vendor=Oracle Corporation 
java.vm.specification.version=11 
java.vm.vendor=BellSoft 
java.vm.version=11.0.15+10-LTS 
jdk.debug=release 
jnlp.jogamp.tmp.cache.root=jln5571435496167399412 
line.separator=\u000d\u000a 
os.arch=amd64 
os.name=Windows 11 
os.version=10.0 
path.separator=; 
sun.arch.data.model=64 
sun.awt.enableExtraMouseButtons=true 
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler 
sun.boot.library.path=c:\program files\openrocket\jre\bin 
sun.cpu.endian=little 
sun.cpu.isalist=amd64 
[... 119 lines cut ...]
332 182.890 VBOSE (DoubleModel.java:914) DoubleModel[constant=0.2] adding listener (total 1): net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@67d01e6b 
333 182.890 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderLength] adding listener (total 2): javax.swing.JSlider$ModelListener@2e33b8ea 
334 182.890 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderLength] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@5b699d0e 
335 182.890 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderLength] removing listener (total 2): javax.swing.plaf.basic.BasicSliderUI$Handler@5b699d0e 
336 182.890 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderLength] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@42100794 
337 182.890 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@1850c72, current number of listeners is 40 
338 182.890 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@436b9771, current number of listeners is 41 
339 182.891 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderThickness] adding listener (total 1): javax.swing.JSpinner$ModelListener@45ee2e55 
340 182.891 VBOSE (DoubleModel.java:914) DoubleModel[constant=0.0] adding listener (total 2): net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@7ee2fdf8 
341 182.891 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderRadius] adding listener (total 1): net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@7ee2fdf8 
342 182.891 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderThickness] adding listener (total 2): javax.swing.JSlider$ModelListener@4acae8f7 
343 182.891 VBOSE (DoubleModel.java:914) DoubleModel[NoseCone:AftShoulderThickness] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@1098ac2a 
344 182.891 VBOSE (DoubleModel.java:936) DoubleModel[NoseCone:AftShoulderThickness] removing listener (total 2): javax.swing.plaf.basic.BasicSliderUI$Handler@1098ac2a 
[... cut ...]
350 185.651 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 2,36480 user scale, 8937,83 overall scale 
351 185.651 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 2,36480 user scale, 8937,83 overall scale 
352 185.652 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 2,36480 user scale, 8937,83 overall scale 
353 189.564 USER (BasicFrame.java:882) Bug report selected 
354 189.565 DEBUG (JarUtil.java:40) Found jar file using codeSource 
---------- End of bug report ---------- 
[image]

Skyfinder84, 2022-09-21:
The same issue is present IN STABLE VERSION 15, not only in the beta. THe beta is been a try, but same issue. Tried to change different graphic settings, no way.

artao5, 2022-10-14:
I had this problem yesterday after doing a bunch of simulations, and I'd been in the software quite awhile (a couple hours or so). Closing and re-opening fixed it. I am not sure how to reproduce it. If I can I'll post the bug report. Just chiming in as also having this problem.
I am using an AMD Radeon RX580 and see no options to change the rendering device.

SiboVG, 2022-10-14:
> (v22.02.beta5) I had this problem yesterday after doing a bunch of simulations, and I'd been in the software quite awhile (a couple hours or so). Closing and re-opening fixed it. I am not sure how to reproduce it. If I can I'll post the bug report. Just chiming in as also having this problem. I am using an AMD Radeon RX580 and see no options to change the rendering device.

Sounds like a memory issue. OR is known to take up a lot of memory. This can pile up over time (due to insufficient memory cleaning within the program), and cause your program to crash...

Skyfinder84, 2022-10-22:
Anyway, is STILL A BUG not fixed...

SiboVG, 2022-10-22:
> Anyway, is STILL A BUG not fixed...

We are aware.

jet689, 2024-02-18:
I am getting this problem still

Phyronian, 2024-08-20:
I noticed something new with this error just now.  The visual errors only begin when I go to change the units for a given input for a part.  Without triggering the issue, I can change a nosecone shape, material, finish, etc. Still, the error immediately occurs if I try and change the units for any of the other inputs.  This also occurs on other parts, so nose cones are not unique.

jet689, 2024-08-20:
I am pretty sure that it doesn’t happen when ran in administrator mode. Try it on your own because if it does there is a way to get the app to automatically ask to be ran in administrator mode.

Phyronian, 2024-08-20:
Hmm, yeah that does seem to work.  Very weird workaround, but I'm not complaining lmao.

SiboVG, 2024-08-22:
Thanks for your reports @jet689 and @Phyronian! See also [this TRF post](https://www.rocketryforum.com/threads/announcement-openrocket-23-09-is-now-available-for-download.183186/post-2624806), where a user had to go to OpenRocket's program properties, click on Run as Administrator AND click on Change high DPI settings then check the "Use this setting to fix scaling problems for this program instead of the one in Settings".
[image]

SiboVG, 2024-08-22:
@jet689 or @Phyronian would any of you be willing to help me fix this bug natively? I have an idea that could fix this issue, without the Administrator hack, but I have no way of testing it, since I can't recreate this bug on my machine.

It would just involve me sending you a JAR file with a potential fix, and you testing it and comparing it with the OpenRocket.23.09.JAR file.

PR #2723 (2025-02-23): [#1667] Eliminate window ghosting on Windows!!
This PR fixes #1667, the infamous window ghosting bug for Windows users. The fix involves adding these JVM arguments when running the JAR, as taken from [here](https://github.com/skylot/jadx/issues/2225):
```
"-Dsun.java2d.noddraw=true"
"-Dsun.java2d.d3d=false"
"-Dsun.java2d.ddforcevram=true"
"-Dsun.java2d.ddblit=false"
"-Dswing.useflipBufferStrategy=True" 
```

The fix has been confirmed by users CoreCount and Catalyst on the r/rocketry server, thanks for the help!

Today is a good day :)
closed 2025-02-23 by SiboVG