openrocket/openrocket #1668: Centering thickness is incorrect.

sjh114961, 2022-09-12:
I reported to Dave cook, "Thickness of CR5060-W should be 0.125 NOT 0.05. Balsa Machining part". His response was "Sounds like a plausible error but where are you seeing this? I've not included a BMS file in my database (yet) due to a bunch of missing manufacturer specs for various balsa parts. If you're looking in the old 15.03 parts source code, none of that code has been carried forward to the new release directly." Is this true or do you guys need to fix this? Dave's github issue https://github.com/dbcook/openrocket-database/issues/13

JoePfeiffer, 2022-09-12:
@dbcook was incorrect on that -- the old database is still available, since it contains parts Daves's doesn't (like BMS!).  You can control whether or not use see legacy parts by opening the Parts Library dialog and clicking the "Show Legacy Database" checkbox.

sjh114961, 2022-09-12:
According to OR this is NOT Legacy  
[image]

sjh114961, 2022-09-12:
So something is wrong

dbcook, 2022-09-12:
> @dbcook was incorrect on that -- the old database is still available, since it contains parts Daves's doesn't (like BMS!). You can control whether or not you see legacy parts by opening the Parts Library dialog and clicking the "Show Legacy Database" checkbox.

I stand corrected on that, though I strongly recommend against keeping or using the legacy database due to the massive number of errors it contains (several hundred before I stopped counting).  I would like to enable the total removal of the old database because of the low fidelity, age (the data set is a decade old) and lack of substantiation.

It seems like there's enough interest that I'll start on a BMS file with whatever parts can be substantiated - CRs, tubes and some balsa parts.  I can't say how long that will take since the primary dev computer I use has died with hardware issues and I'm not sure it will come back to me with the hard drive intact with my "special" installations of OR, vscode XML tools, etc.  It will be a week and maybe two before I can do anything that requires serious testing.  In the meantime I can make simple textual fixes only.

@JoePfeiffer , the non-legacy designation for the BMS parts looks wrong.  Not sure exactly what the bug is since I'm not the maintainer of the logic that decides what's legacy or not.  I just confirmed that there are no parts in any file in my DB repo with mfr containing "BalsaMachining", so I've not accidentally included the bms file in my own repo.  It has to be either a packaging problem or a logic flaw upstream in OR.

sjh114961, 2022-09-12:
I am retired and have some free time. If I can help in anyway, just ask. I have added componet of my own to OR so I am not totally blind. Perhaps another manufacturer you wish is include

JoePfeiffer, 2022-09-13:
Yeah.... it took me a minute to remember what I did there.  Manufacturers for which we don't have newer versions from Dave aren't marked as legacy, since they're the best we've got.  It didn't didn't occur to me that users would (quite reasonably, in retrospect) conclude anything not legacy was Dave's.  I've just put a note on the dev-questions slack channel; if I don't hear *really* strunious objections I'll be marking all the stuff from 15.03 as "legacy".

I'm torn on just eliminating the 15.03 database.  I really want to do it eventually, but for the moment that's the only place to find quite a few parts.  It's got *many* errors, but it's not clear to me that it's got so many that it does more harm than good.

sjh114961, 2022-09-13:
In my opinion, I like the way you have it now, however, I would like all the NON-Dave stuff marked as legacy.

hcraigmiller, 2022-09-13:
The new parachutes and rail buttons should NOT be marked as legacy.

JoePfeiffer, 2022-09-13:
> The new parachutes and rail buttons should NOT be marked as legacy, even though not Dave’s.

Thanks for reminding me -- I'd have forgotten.

closed 2022-09-17 by JoePfeiffer

dbcook, 2022-10-11:
Yes, the non-legacy designation for the BMS parts looks wrong.  Not sure
exactly what the bug is since I'm not the maintainer of the logic that
decides what's legacy or not.  I just confirmed that there are no parts in
any file in my DB repo with mfr containing "BalsaMachining", so I've not
accidentally included the bms file in my own repo.  It has to be either a
packaging problem or a logic flaw upstream in OR.

JoePfeiffer, 2022-10-11:
> Yes, the non-legacy designation for the BMS parts looks wrong. Not sure exactly what the bug is since I'm not the maintainer of the logic that decides what's legacy or not. I just confirmed that [...]

I think that was changed in the PR referenced above.  Regardless, BMS is marked as legacy in Beta 5 (as are all parts from 15.03).

PR #1669 (2022-09-12): Correct thickness of BMS CR5060-W centering ring.
fixes #1668