openrocket/openrocket #1673: [Feature Request] Data for booster stage(s) in simulation list

andrewjbare, 2022-09-13:
My team is designing a two-stage rocket for the first time and we're interested in ground hit velocity and apogee for the booster stage. Although we can get this data from plot/export, for iterative design it would be nice to see this data right in the simulation window. I think just a few more columns per stage ought to do the trick. 
With more data available it would be good to choose what columns are shown in the simulation list (i.e. if I don't care all that much about time to apogee, I shouldn't have to see it). I'm not sure what the most elegant way to do this is, but it would be a nice QoL upgrade for our iterative design.

SiboVG, 2022-09-13:
Could be an interesting feature! Although chances are slim that it will be included in OR any time soon.

JoePfeiffer, 2022-09-13:
I'd more generally like the user to be able to select whatever columns they want in all of our data tables (simulation data, component library selection, component analysis....).  It's going to be high on my wish list once this release is out.

SiboVG, 2025-02-05:
Superseded by #2704.

closed 2025-02-05 by SiboVG