openrocket/openrocket #1679: [Bug] Freeform fin missing in 3D views

hcraigmiller, 2022-09-18:
I'm not sure what is causing this. . . I even copied the 13mm hook to the 24mm mount, and it still disappeared.

[image]

[Nested.Estes 24mm Motor Mount.No Error.txt](https://github.com/openrocket/openrocket/files/9592882/Nested.Estes.24mm.Motor.Mount.No.Error.txt)
[Nested.Estes 13mm Motor Mount.Error.txt](https://github.com/openrocket/openrocket/files/9592883/Nested.Estes.13mm.Motor.Mount.Error.txt)

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

hcraigmiller, 2022-09-18:
I figured it out. . . the starting and ending points cannot both be 0,0.

SiboVG, 2022-11-17:
> I figured it out. . . the starting and ending freeform fin points cannot both be 0,0; need a warning in the Wiki.

Actually, OR should be able to handle such an "error". So I'm going to re-open this issue.

PR #1859 (2022-12-04): [#1679] Render fin set even if zero length
This PR fixes #1679 by still rendering the fin set even if it has a zero-length (which happens when the last fin point is overlapping with the first one).
closed 2022-12-04 by SiboVG