openrocket/openrocket #1683: Pre-selected transition only work in 1 direction

sjh114961, 2022-09-20:
This is a copy of an issue entered in the parts database.

Transitions work in only 1 direction, generally smaller to larger tube. There is no way currently in OR to flip it to the other direction, Say a transition from a BT20 to a BT50, to the other direction BT50 to BT 20.

Solutions:
1) Get the OR Devs to provide a switch for the direction. 
2) Duplicate all transition in data base with the reversed parameters but add only a "(R)" to show that the direction is reversed. This is far more cumbersome and affect all manufactures.

Since nosecones can also be used as tailcones, they have the same problem.

SiboVG, 2022-09-20:
Option 1 has my preference, option 2 would clutter the database and confuse users.

A more elaborate option could be to add a mirror transform option in OR, although it would probably only be useful for transitions and fin sets.

sjh114961, 2022-09-20:
AND nosecones being used as tailcones

SiboVG, 2022-09-20:
> AND nosecones being used as tailcones

But OR doesn't support reversed nosecones, right? A "reversed nosecone" would simply be a transition with the aft diameter set to zero.

sjh114961, 2022-09-20:
Correct

JoePfeiffer, 2022-09-20:
> > AND nosecones being used as tailcones
> 
> But OR doesn't support reversed nosecones, right? A "reversed nosecone" would simply be a transition with the aft diameter set to zero.

Yes, but he's talking about components from the library. It ought to be possible to put a reversed nose cone at the end of a body tube for cool looking pods and things.

dbcook, 2022-09-20:
Agree, a "reverse" switch would be quite handy.  For nose cone reversal we'd need to make sure that the Barrowman engine could estimate the CP, inertia, and Cd contributions reasonably, but I don't think that would be a big problem.  There are around 100 transitions already in the parts database with [R] reversed versions, which is handy because without them you have to create reversed transitions manually by looking up the ID/OD of both tube sizes.

neilweinstock, 2022-12-15:
Since I just inadvertently submitted a duplicate request, I’ll chime in here and add my request for this feature. 

As for modeling backwards nose cones etc, we already support the exact same set of shapes in transitions; this feature is a very useful UI convenience, not anything new with regard to modeling.