openrocket/openrocket #1685: Edit Window

MGD713, 2022-09-22:


MGD713, 2022-09-22:
Hey everyone, I have been using OR for some time now and recently I cannot open the edit window. I am under the impression this issue is pretty common and is caused when you connect to a monitor and it opens in a "ghost window" I have looked all over and cannot find a solution. I am using 22.04 beta 4. Any help would be greatly appreciated!

SiboVG, 2022-09-22:
You can go over [this post](https://www.reddit.com/r/rocketry/comments/s39c1k/openrocket_edit_window_does_not_open/?utm_source=share&utm_medium=web2x&context=3) or [this post](https://www.reddit.com/r/rocketry/comments/m1e6tn/cant_edit_parts_in_openrocket/) to find a temporary solution for your problem.

MGD713, 2022-09-22:
Thanks, unfortunately none of those solutions have worked so far. When I create the body tube (as in #1475), the config dialog does not pop up. Any ideas?

SiboVG, 2022-09-22:
> Thanks, unfortunately none of those solutions have worked so far. When I create the body tube (as in #1475), the config dialog does not pop up. Any ideas?

The steps in #1475 are not for fixing your issue, but are steps of testing the code I wrote for fixing the issue.

If you're familiar with running JAR files, you can try running [this jar](https://github.com/openrocket/openrocket/suites/8373328973/artifacts/369417944), it contains the fix for your issue. I'm not sure, but by running OpenRocket from that JAR, it may also fix the issue in the OpenRocket application in which you're experiencing the issue.

MGD713, 2022-09-22:
Ok, I tried running the JAR file but it says its unsupported.
Error:
Unsuppored Java versionL 1.8.0_311
Supported version(s): 11, 17
Please change the Java Runtime Environment version or install OpenRocket using a packaged installer

Thank you for helping me

SiboVG, 2022-09-22:
> Ok, I tried running the JAR file but it says its unsupported. Error: Unsuppored Java versionL 1.8.0_311 Supported version(s): 11, 17 Please change the Java Runtime Environment version or install OpenRocket using a packaged installer
> 
> Thank you for helping me

You should download Java 11 for your operating system [here](https://www.oracle.com/be/java/technologies/javase/jdk11-archive-downloads.html#:~:text=Java%20SE%20Development%20Kit%2011.0.16).

MGD713, 2022-09-22:
It worked! Thank you so much!

SiboVG, 2022-09-22:
> It worked! Thank you so much!

Great :) enjoy!