openrocket/openrocket #1690: Write unit tests for multi-component editing

SiboVG, 2022-09-23:
Would be good to write some unit tests on multi-component editing (e.g. a test that checks that if you have a design with two body tubes, select both of them and then change the length of the body tube, whether the length change is applied to both body tubes).

Alisonsun7, 2026-10-13:
Hi @SiboVG, I would like to work on this and could you assign it to me?

JoePfeiffer, 2026-10-13:
I'm a little puzzled: I got a notification that @AaronJin0323 also wished to work on this issue, but I'm not seeing that comment now.

We're more than happy to accept contributions from members of a class, but we'd really appreciate it if you could coordinate among yourselves which issues you would like to help with before volunteering so we aren't trying to do that coordinattion.

Alisonsun7, 2026-10-14:
Hi @JoePfeiffer,

Sorry for the confusion. Initially, Aaron planned to comment on behalf of our entire group to express our interest in contributing. However, we later realized that this might cause confusion about who would be working on each issue, so Aaron deleted his earlier comments.
We have since discussed this within our group and decided that each member will comment individually on the specific issue they plan to work on.

Thank you for your feedback!