openrocket/openrocket #1694: Apply change in stability unit without having to reopen design

SiboVG, 2022-09-24:
Small QoL change: when changing the stability unit in the preferences, it would be nice to not have to close the design file, and reopen it to apply the unit change. Instead, after changing the stability unit, the change should take effect in the rocket view immediately after you close the preference dialog (or possibly even after you change the unit, and still have the preferences dialog open).

JoePfeiffer, 2022-09-25:
This would apply to all the preferences.

hcraigmiller, 2022-10-28:
Functions as expected, no anomalies found. Very nice. 

Build 1133
[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

PR #1772 (2022-10-28): [#1694] Update rocket figure after unit change
This PR fixes #1694. If you change the units in the preference dialog, and close the dialog, the units in the rocket view should get updated.

Note: the units for the stability info and flight info are not updated in the 3D view, see issue #1771.
closed 2022-10-28 by SiboVG