openrocket/openrocket #1708: 3D views and Photo Studio issues

smstachwick, 2022-09-30:
After upgrading to Public Beta 5, I've observed issues in the 3D views that carry over to Photo Studio.  Specifically, the rendering appears to be misplaced on my screen.

[image]

[image]

The right edge of the 3D Finished view is cut off.  Also note that CP and CG are off-centerline, which was also the case in Public Beta 4.  Both issues are present in 3D Unfinished view.

hcraigmiller, 2022-09-30:
This appears to be a settings issue on your device. Please follow these steps and reports back: http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings

smstachwick, 2022-10-09:
> 

That fixed it.  Please feel free to close this issue.

closed 2022-10-09 by SiboVG