openrocket/openrocket #1728: [Bug] Uncaught exception when adding/running new simulation

hcraigmiller, 2022-10-09:
With run outdated simulations on:

1. Open .ork
2. Run simulation with existing configuration.
3. Add second configutation (same one); runs very slow.
4. Go to flight simulations tab.
5. Add third configutation (same one)
4. Go to flight simulations tab (uncaught exception).

[image]
[Bug Report.txt](https://github.com/openrocket/openrocket/files/9742769/Bug.Report.txt)
[Sea Killer.ork.txt](https://github.com/openrocket/openrocket/files/9742770/Sea.Killer.ork.txt)

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-10-10:
I'm kindly handing this bug over to @JoePfeiffer, it's a bit over my head. For some reason, `status.getRocketPosition().z` on line 443 in `swing/src/net/sf/openrocket/gui/simulation/SimulationRunDialog.java` can equal zero, thus setting `apogeeAltitude` to zero, and later triggering the exception on line 368 of `SimulationRunDialog.java`, because you're doing `MathUtil.map(status.getRocketPosition().z, apogeeAltitude, 0, APOGEE_PROGRESS, 1.0)`, which becomes `MathUtil.map(status.getRocketPosition().z, 0, 0, APOGEE_PROGRESS, 1.0)` (fromMin and fromMax in MathUtil.map both become 0).

JoePfeiffer, 2022-10-10:
In addition to that, I'm seeing a couple of other things:
1) Simulations 2 and 3 are coming off the rail much faster than Simulation 1
2) If Simulation 3 hits the ground before the exception happens, the ground hit velocity is ridiculous
Are you also seeing these?
[image]

SiboVG, 2022-10-10:
> In addition to that, I'm seeing a couple of other things:
> 
> 1. Simulations 2 and 3 are coming off the rail much faster than Simulation 1
> 2. If Simulation 3 hits the ground before the exception happens, the ground hit velocity is ridiculous
>    Are you also seeing these?
>    [image]

The first one not, just because I wasn't monitoring that, but I did noticed the second issue as well.

JoePfeiffer, 2022-10-11:
The velocity off the rod was a red herring -- when duplicating a configuration, the simulation conditions aren't duplicated so my simulations 2 and 3 were using a 17 foot rail.

SiboVG, 2022-10-23:
I have to reopen this. #1755 did not fix this issue. In fact, strangely, it appears that I can more easily replicate this issue now.

1. Open 'A simple model rocket'
2. Select all the simulation
3. Click "Run simulations"
4. First run goes well
5. Click "Run simulations" again
6. Exception

[bug report.txt](https://github.com/openrocket/openrocket/files/9847629/bug.report.txt)

Pinging @JoePfeiffer so he doesn't miss this message.

JoePfeiffer, 2022-10-23:
Thanks for the ping.

PR #1755 (2022-10-23): Improve progress bar reporting
The progress bar dialog was trying to keep track of simulation apogee in two places; in one of them the "apogee" was actually one simulation status update after apogee.  Combined with the bug fixed in PR #1749, this could lead to a reported apogee of 0, causing the exception reported in #1728.  This PR removes the redundant apogee calculation.

Fixes #1728

The SimulationRunDialog sitll needs significant cleanup, but I'd like to just get this bug fix in now and save that for a later PR.
PR #1765 (2022-10-26): Improve sampling for SimulationRunDialog
It turned out there were two (more) issues:

1. The simulation ran so fast the rocket was on the ground before the first progress update happened.  Since the SimulationStatus was simply being published, this meant that all of the entries in the publish queue were references to the same status, and all the samples before it were effectively lost. Publishing a copy of the SimulationStatus fixes this.

2. Since vertical velocity is slightly less than 0 when apogee is detected, the max altitude code wasn't picking up on the apogee event. Looking for either  [... cut]
closed 2022-10-26 by SiboVG