openrocket/openrocket #1729: Adding more motors to the database (feature request)

Nighten7, 2022-10-10:
I'd like to see some more motors in the database. I was working on one of my rockets the other day and noticed that the Aerotech H219T and I175W were not present. if you could add some more high power motors (particularly L1 in my case, but others would be nice as well) that would be great. maybe you could use thrustcurve.org to get the information necessary to run simulations with a motor?

SiboVG, 2022-10-10:
We use thrustcurve.org for our motor database. Both the H219T and I175W are present, but you need to search them using their common name from thrustcurve.org (H219 and I175). I thought the database search box also worked for searching based on the common name, but apparently not, pinging @JoePfeiffer for that.

closed 2022-10-10 by SiboVG

JoePfeiffer, 2022-10-10:
Both of those motors are in our database.  Some things to look at:
1) Don't include the T in H219T or the W in I175W.  For consistency across manufacturers we aren't currently showing the propellant type code; it's been requested that we add that back in, and we will (soon...). I hadn't realized until now that including it in the search would make the search fail -- though while you're typing the motor name in the search box you should see the motor until you reach the propellant designation.
2) Make sure that if you're planning on the motor extending out the front of your motor mount you don't have the "Limit motor length to mount length" box checked.
3) What version of OpenRocket are you running? The H219 wasn't in 15.03 (though the I175 was, so this is unlikely to be it).