openrocket/openrocket #1730: Windows: Copy/paste doesn't work across multiple instances of OR

neilweinstock, 2022-10-10:
See discussion starting here: https://www.rocketryforum.com/threads/adding-parts-to-the-openrocket-database.163446/post-2328410.  Craig figured out how to explain the behavior (https://www.rocketryforum.com/threads/adding-parts-to-the-openrocket-database.163446/post-2331308) but it doesn't really make sense to me.

Essentially: it seems that you cannot copy/paste between different running instances of OR on the same machine.  Original report (linked above) concerned trying to copy/paste between two different beta versions.  Craig also figured out that if you launch OR by double-clicking a file it will launch a new instance of OR rather than opening the file in an existing instance of OR.

I don't know why this limitation should exist.  I further don't know why double-clicking an ORK file creates a new instance rather than opening in an existing instance... perhaps that should be a separate filed issue, I don't know.

The second-instance problem should not exist on Mac (I need to double-check), don't know about Linux.  The issue of copy/pasting between different versions might also exist on Mac.

JoePfeiffer, 2022-10-10:
At least under Linux, the default behavior when you run a program is to run a new instance rather than to look for an existing instance and hook up to it.  Consequently, very few Linux programs will cut-and-paste across instances as being requested here.  There are few (LibreOffice and Gimp come to mind) but they are decided exceptions.  It would take at least a little effort to make OR do this; I'd suggest this be another one that should be high on the list for "after 22.02 final" comes out.

Trying to do it across versions is probably hopeless.  

An intermediate approach that could work (again, under Linux) would be to copy and past to/from what's called "the clipboard"; that would work across versions as long as the data structures haven't changed too much.  My guess at present is that implementing the "look for a running OR and hook up to it" would do a better job of satisfying our users than this.  Again, look seriously after final release of this version.

neilweinstock, 2022-10-10:
Wait, are we not using the clipboard???

SiboVG, 2022-10-10:
> Wait, are we not using the clipboard???

We use an in-program clipboard. Copying to the system clipboard is very bothersome...

neilweinstock, 2022-10-10:
Hmm, disappointing.  Not what users expect.

Failing that, we should try to do what Joe said, which is minimize the likelihood of multiple instances of the same version.  Doesn't help folks copying between versions.  (although I'm not quite sure why that would be necessary, to be honest...)  I understand if that can't make it into this release.

neilweinstock, 2022-10-12:
I've changed this issue to "use the system clipboard", since that is what is needed to address the original issue.  If we can't do this for technical reasons right now that's fine.

I will file a separate issue to address the "don't open multiple instances of OR" problem.