openrocket/openrocket #1737: Windows: try not to open multiple instances of OR

neilweinstock, 2022-10-12:
On Windows, if you double-click an ORK file in Explorer, then a new instance of OR will be created to open that file, even if there is an instance of OR already running.  This is undesirable behavior: since OR can handle multiple open designs at once, we should strive to keep only a single instance running at once.  In addition to saving system resources and generally avoiding confusion, it also ensures that copy/paste will work between designs, at least until the system clipboard is used (see also #1730.)

And so, when double-clicking an ORK file, it should ideally open within an existing instance if one exists.  I believe this behavior is already the norm on MacOS but it needs to be verified (that would be normal expected behavior on Mac, where there's really no such thing as having multiple instances of the same program running).

I am unsure of Linux behavior.

Originally discussed on TRF: https://www.rocketryforum.com/threads/adding-parts-to-the-openrocket-database.163446/post-2328410

JoePfeiffer, 2022-10-15:
This is as valid a request for Linux as it is for Windows.

SiboVG, 2023-01-11:
Looks like install4j has such a feature built-in, under the Launcher options.
[image]

Will have to test it out later.

SiboVG, 2023-01-12:
> Looks like install4j has such a feature built-in, under the Launcher options.

Nevermind, just tested it, it's not as simple as checking a checkbox 🙂. The single instance mode allows only one design file to be open at a time. We also need to implement some code to fix this issue.

senft-research, 2025-05-05:
Me and another developer will be having a look at this after this week is concluded, once current commitments are satisfied.