openrocket/openrocket #1775: Bugs: Negative "Optimum Delay" on 2 stage rocket & parachute does nothing (22.02.beta.05)

Nighten7, 2022-10-28:
[image]
so 3 problems here:
1) the optimum delay for ejection is negative
2) the rocket hits the ground at over 170 mph using a parachute that is actually a little oversized for a rocket of this weight
3) velocity at deployment is nonexistent (probably related to problem 1)

the velocity off rod values are also incorrect just because I forgot to change the rod length to my longer rod in the sim before taking the screenshot.

sorry if this issue has already been reported but I didn't see anything too similar to it so I figured this may be useful to send in.
thanks,
James

JoePfeiffer, 2022-10-28:
Please post the .ork file for the rocket.  You'll need to zip it or change its name to file.ork.txt to get github to let you upload it.

SiboVG, 2022-11-06:
@Nighten7 could you post your .ork file so we can debug this issue?

JoePfeiffer, 2022-11-07:
You're using delays on your booster motors that cause second stage ignition to happen after apogee (so your rocket is being powered into the ground).  Consequently, the calculated optimal delay is before motor burnout. Use 0-delay booster motors and you'll get more reasonable results.

closed 2022-11-07 by JoePfeiffer

Nighten7, 2022-11-07:
oh. well that's a little embarrassing. sorry for the trouble then.

JoePfeiffer, 2022-11-07:
No problem.  Happy to look at it.