openrocket/openrocket #1777: NaN errors opening .ork files which have motor configurations populated

questionablerocket, 2022-10-30:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. In OpenRocket build from fresh source at 2022-10-30 22:02 UTC, attempting to open any rocket which has motor configurations defined results in NaN errors on motor diameter and length.  In this case, I tried to open the sample rocket "A simple model rocket"

2.  A new rocket can be created instead, but if it is saved with a motor installed, its file cannot subsequently be opened.  If motor configurations are deleted prior to saving, then a newly created rocket can be reopened.  Motor configurations can't be removed from existing rockets populated with motors, since they cannot be opened.

3. Tested on Ubuntu 18.04 and Debian 11.  In the later case, OpenRocket had not been run on the machine before, so there were no defined motor files configured and nothing lurking in the user home directory.

4. The previous build I made from fresh source at 2022-10-24 17:10 UTC did not have this problem.  I have not yet had the time to go back through the commits in the time range between working and non-working versions to isolate at what commit the behavior began.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.util.BugException: BUG: Unknown error when opening file
at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1263)
at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1198)
at net.sf.openrocket.gui.main.ExampleDesignFileAction$1.actionPerformed(ExampleDesignFileAction.java:60)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
at java.desktop/java.awt.Component.processEvent(Component.java:6391)
at java.desktop/java.awt.Container.processEvent(Container.java:2266)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.util.concurrent.ExecutionException: java.lang.NoSuchMethodError: 'net.sf.openrocket.rocketcomponent.FlightConfigurationId net.sf.openrocket.rocketcomponent.Rocket.createFlightConfiguration(net.sf.openrocket.rocketcomponent.FlightConfigurationId)'
at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
at java.desktop/javax.swing.SwingWorker.get(SwingWorker.java:613)
at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1238)
... 39 more
Caused by: java.lang.NoSuchMethodError: 'net.sf.openrocket.rocketcomponent.FlightConfigurationId net.sf.openrocket.rocketcomponent.Rocket.createFlightConfiguration(net.sf.openrocket.rocketcomponent.FlightConfigurationId)'
at net.sf.openrocket.file.openrocket.importt.MotorMountHandler.closeElement(MotorMountHandler.java:80)
at net.sf.openrocket.file.simplesax.DelegatorHandler.endElement(DelegatorHandler.java:100)
at java.xml/com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.endElement(AbstractSAXParser.java:618)
at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanEndElement(XMLDocumentFragmentScannerImpl.java:1728)
at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:2899)
[... 322 lines cut ...]
255 10.183 STDERR (SwingExceptionHandler.java:49) at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:889)
256 10.183 STDERR (SwingExceptionHandler.java:49) at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:825)
257 10.183 STDERR (SwingExceptionHandler.java:49) at java.xml/com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141)
258 10.183 STDERR (SwingExceptionHandler.java:49) at java.xml/com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.parse(AbstractSAXParser.java:1224)
259 10.183 STDERR (SwingExceptionHandler.java:49) at java.xml/com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl$JAXPSAXParser.parse(SAXParserImpl.java:637)
260 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.simplesax.SimpleSAX.readXML(SimpleSAX.java:51)
261 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.openrocket.importt.OpenRocketLoader.loadFromStream(OpenRocketLoader.java:51)
262 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.AbstractRocketLoader.load(AbstractRocketLoader.java:21)
263 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.GeneralRocketLoader.loadUsing(GeneralRocketLoader.java:240)
264 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.GeneralRocketLoader.loadRocket(GeneralRocketLoader.java:201)
265 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.GeneralRocketLoader.loadStep1(GeneralRocketLoader.java:160)
266 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.file.GeneralRocketLoader.load(GeneralRocketLoader.java:94)
267 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.util.OpenFileWorker.doInBackground(OpenFileWorker.java:72)
268 10.183 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.util.OpenFileWorker.doInBackground(OpenFileWorker.java:28)
269 10.183 STDERR (SwingExceptionHandler.java:49) at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
270 10.183 STDERR (SwingExceptionHandler.java:49) at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[... cut ...]
273 10.183 STDERR (SwingExceptionHandler.java:49) at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
274 10.183 STDERR (SwingExceptionHandler.java:49) at java.base/java.lang.Thread.run(Thread.java:833)
275 10.183 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
276 10.183 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
277 12.498 USER (SwingExceptionHandler.java:220) User requested sending bug report
278 12.521 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

questionablerocket, 2022-10-30:
I isolated this to the following commit:

https://github.com/openrocket/openrocket/commit/be0457e9801dcfd82a958ccf921e58071a276781

When I build from the previous commit, 9df0fd9bb6dd8c94425524ce4d4acbadcd361104 I do not have the problem.  When I build with be0457e9801dcfd82a958ccf921e58071a276781 I do.

SiboVG, 2022-10-30:
Did you try running `ant clean` before building? I can't replicate this issue and remember having the same issue as you, which I solved by cleaning my local build.

questionablerocket, 2022-10-30:
Sibo, you are correct.  After running ant clean, I no longer have the issue.  Closing. Thank you.

closed 2022-10-30 by questionablerocket

SiboVG, 2022-10-30:
> Sibo, you are correct. After running ant clean, I no longer have the issue. Closing. Thank you.

Alright, no problem, happy to help 👍