openrocket/openrocket #1778: ork file causes stack overflow

JoePfeiffer, 2022-10-31:
User reports importing rocksim file, after some editing OR crashes.  The user wound up with the attached .ork file, which causes a stack overflow (and no stack trace) when I try to load it.
[LOC-75 edited.ork.txt](https://github.com/openrocket/openrocket/files/9901767/LOC-75.edited.ork.txt)

SiboVG, 2022-10-31:
The issue is caused by the parachute. If you delete that component from the .ork, the error disappears. Still searching for the exact cause.

SiboVG, 2022-10-31:
*ding, ding* found it. The culprit is the following line from the parachute's XML:
```XML
<preset type="PARACHUTE" manufacturer="LOC/Precision" partno="LP-78" digest="eb283078de76260abab77feb91810714"/>
```

SiboVG, 2022-10-31:
Okay, the issue was with the LP-78 parachute preset. There was a bug on importing parachutes (#1723), which was solved in #1726. But the issue is already solved, so closing this.

closed 2022-10-31 by SiboVG