openrocket/openrocket #1790: [Bug] Wrong simulation results from TARC Payloader example rocket

SiboVG, 2022-11-04:
Bug report from Chris Flanigan on the SourceForge bug mailing list:

---------- Bug report ----------

*Include detailed steps on how to trigger the bug:*

*(You can edit text directly in this window)*

1. start OpenRocket

2.  "File, Open Example, TARC Payloader"

3.  Select "Flight simulations" tab

4.  "Plot/Export, Plot" (to display the saved simulation results)

5.  Close the plot windows

6.  "Run Simulations" (to recalculate the simulation results)

7.  "Plot/Export, Plot" (to display the newly calculated simulation results)

The new results do not match the saved results.  It appears that the
parachute for the "Payload" never gets activated.
*What does the software do and what in your opinion should it do in the
case described above:*

The TARC Payloader example indicates that something about upper stage
parachutes isn't working.  The results plotted after doing "Run
simulations" doesn't match the original results.  The results also don't
match the OpenRocket v15 results for the TARC Payloader example.

If you designate the "Body tube" of the "Payload" stage as a motor mount,
add a tiny motor with a short delay (such as 1/4A3 with 0.1 second delay),
and change the "Payload" parachute to "First ejection charge of this
stage", the results still don't look correct.  It looks like the "Payload"
stage is using the delay time from the booster motor.

Include your email address (optional; it helps if we can contact you in
case we need additional information):

chris...@gm...

(Do not modify anything below this line.)

---------- System information ----------

OpenRocket version: 22.02.beta.05

OpenRocket source: default

OpenRocket location:
C:\Users\cflanigan\AppData\Local\OpenRocket\.\OpenRocket.jar

JOGL version: 2.4.0-rc-20210111

Current default locale: en_US

System properties:

awt.toolkit=sun.awt.windows.WToolkit

exe4j.consoleCodepage=cp0

exe4j.isInstall4j=true

exe4j.launchName=C:\Users\cflanigan\AppData\Local\OpenRocket\OpenRocket.exe

exe4j.moduleName=C:\Users\cflanigan\AppData\Local\OpenRocket\OpenRocket.exe

[... 706 lines cut ...]
[javax.swing.JSlider$ModelListener@30b9d590,
javax.swing.plaf.basic.BasicSliderUI$Handler@2d8a1dc0]

8010 256.211 DEBUG (MemoryManagement.java:56) Adding object into
collectable list: DoubleModel[SimulationOptions:TimeStep]

8011 256.211 VBOSE (DoubleModel.java:942)
DoubleModel[SimulationOptions:TimeStep] removing listener (total 0):
javax.swing.JSlider$ModelListener@30b9d590

8012 256.211 VBOSE (DoubleModel.java:953) Invalidating
DoubleModel[SimulationOptions:TimeStep]

8013 256.211 DEBUG (MemoryManagement.java:56) Adding object into
collectable list: DoubleModel[SimulationOptions:TimeStep]

8014 256.222 DEBUG (MemoryManagement.java:56) Adding object into
collectable list:
net.sf.openrocket.gui.simulation.SimulationEditDialog[dialog6,26,26,674x520,invalid,hidden,layout=java.awt.BorderLayout,DOCUMENT_MODAL,title=Plot
/ Export
simulation,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,7,30,660x483,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]

8015 454.165 DEBUG (GUIUti...
 
[truncated message content]

hcraigmiller, 2022-11-04:
With regard to the payload parachute, it appears that a deployment of lower separation at "0" seconds results in the error (the parachute never deploys). Increasing the deployment of lower separation at greater than "0" gives the correct result.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

JoePfeiffer, 2022-11-05:
Please upload the .ork to this bug report

JoePfeiffer, 2022-11-05:
For some reason, when the RECOVERY_DEVICE_DEPLOYMENT event is processed a TUMBLE event is being inserted into the queue, so it tumbles instead of coming down under chute.  Hmmmm