openrocket/openrocket #1806: Write unit tests for component parameter saving and re-opening

SiboVG, 2022-11-09:
In light of issue #1792, it would be good to include unit testing on saving rocket components, opening that saved file and verifying that all the values are as expected. (this is also not the first time that a similar issue like #1792 happened)

Preferably, the parameters are set to random values, keeping in mind a sufficient range (cf. issue #1792 where you have to test the thickness when it's both smaller and greater than 2.5 cm).

isaacjones99, 2023-11-13:
Hi, can I work on this?

SiboVG, 2023-11-13:
> Hi, can I work on this?

Sure, that'd great!

isaacjones99, 2025-03-17:
Hello,

Sorry it's taken me so long to work on this.

I’ve been looking at this over the weekend, but I’m unsure how to proceed. From what I understand, we need a unit test to verify that values saved to a file are correct when the file is reopened. My challenge is accessing the values once they’ve been loaded into the `Rocket` object. Could you guide me on how to approach this?

So far, I’ve tried loading the `Terrier.Rev_02.Stripped.ork` file mentioned in #1792 into an `OpenRocketDocument`, from which I can retrieve the `Rocket` object. However, I’m not sure how to access or validate the specific values stored within the `Rocket` after it’s been loaded.

Also, given issue #1792, which involved loading the transition wall thickness: should the unit test focus specifically on this property, or should it cover all properties saved to the file?

JoePfeiffer, 2025-03-17:
The Rocket is a tree structure; you can get the children of a component with getChild(n) where n identifies the child.

So, for instance, starting with the Rocket, you can get the sustainer AxialStage with rocket.getChild(0).  If you look at the rockets in core/src/main/java/info/openrocket/core/util/TestRockets.java, you can see how the rockets in that file are constructed with calls to addChild().

Hope this helps,

Astra-Erevos, 2026-10-13:
Hi! My teammate @baizhenh884-dev and I would like to work on this issue, if it is still available.
We are students from ANU working on an open-source contribution assignment, and our tutor has approved this issue for the two of us to work on together.
We’ve read the previous discussion and Joe’s guidance about traversing the Rocket tree using getChild(n) and referring to TestRockets.java. Our current plan is to add unit tests that set component parameters to representative/random values, save the rocket to a file, reopen it, traverse the component tree, and verify that the parameter values are preserved. We’ll also make sure to include relevant boundary cases such as the transition wall thickness case from #1792.
We noticed that this issue had a previous volunteer but currently has no assignee or linked PR. Would it be okay for us to take this on?
Thanks!

JoePfeiffer, 2026-10-13:
It looks like you've also volunteered to take on #1690 and #1692. Please pick the one that looks like the best fit for you, and we'll assign you to it. If your class requires you to make more than one contribution, let's make them one at a time.

Astra-Erevos, 2026-10-16:
> 11\. It looks like you've also volunteered to take on [#1690](https://github.com/openrocket/openrocket/issues/1690) and [#1692](https://github.com/openrocket/openrocket/issues/1692). Please pick the one that looks like the best fit for you, and we'll assign you to it. If your class requires you to make more than one contribution, let's make them one at a time.

Hi @JoePfeiffer, sorry for the confusion. We have already coordinated the work within our group, but I realize we did not clearly explain the assignment of the three issues.
Our agreed division is:
- #1806: @Astra-Erevos and @baizhenh884-dev will work on this together.
- #1690: @Alisonsun7 will work on this.
- #1692: @AaronJin0323 will work on this.
Our tutor has approved this exact division for our assignment: #1806 is treated as a medium-sized issue for two students, while #1690 and #1692 are smaller issues for one student each.
So I personally am only volunteering for #1806, together with @baizhenh884-dev. The comments on #1690 and #1692 are from the other members of our group for their own assigned issues.
Sorry again for the confusion caused by our earlier comments. We would be happy to proceed with #1806 if that works for you. Thank you!