openrocket/openrocket #1807: Don't allow stage separation at launch

JoePfeiffer, 2022-11-10:
Looking at #1611 (Feature request: add warning when no motor selected in first stage of simulation), I discovered that a user can actually leave the booster empty, and specify booster stage separation and second stage ignition both at launch, as shown in the uploaded file.  This should leave the empty booster on the pad while the rest of the rocket launches (trying to plot the resulting flight results in an array out of bounds error that I haven't tried to track down yet, so I can't be entirely sure of that).

But...  can anyone think of any use case in which stage separation occurs at launch? If not, we should remove the option.
[empty-booster.ork.txt](https://github.com/openrocket/openrocket/files/9982182/empty-booster.ork.txt)

hcraigmiller, 2022-11-11:
I am unaware of any situation in which separation occurs or should be allowed to occur at launch.

JoePfeiffer, 2022-12-08:
Closed by #1860

closed 2022-12-08 by JoePfeiffer