openrocket/openrocket #1814: Start up Error

benrussell11, 2022-11-11:
Beta 05 release

I believe this is caused by chute packed length not being handled correctly. The packed length is being recalculated each time the design is loaded. Needless to say the calculated lengths are wrong and keep getting larger.

I'm not sure if this because I'm using chutes from the component database or not.

ben.russell11@gmail.com

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: Row index out of range at java.desktop/javax.swing.JTable.boundRow(JTable.java:2150) at java.desktop/javax.swing.JTable.addRowSelectionInterval(JTable.java:2200) at net.sf.openrocket.gui.main.SimulationPanel.takeTheSpotlight(SimulationPanel.java:1017) at net.sf.openrocket.gui.main.SimulationPanel.deleteSimulation(SimulationPanel.java:356) at net.sf.openrocket.gui.main.SimulationPanel.access$1000(SimulationPanel.java:74) at net.sf.openrocket.gui.main.SimulationPanel$DeleteSimulationAction.actionPerformed(SimulationPanel.java:597) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1020) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1064) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.05
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\ben\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\nodejs\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Users\ben\AppData\Local\Programs\Python\Python37-32\Scripts\;C:\Users\ben\AppData\Local\Programs\Python\Python37-32\;C:\Users\ben\AppData\Roaming\npm;C:\Users\ben\AppData\Local\Microsoft\WindowsApps;C:\Users\ben\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\ben\AppData\Local\GitHubDesktop\bin;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.15+10-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.15
java.version.date=2022-04-19
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.15+10-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln18285968106586938441
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=64
[... 286 lines cut ...]
495732 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
495733 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
495734 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
495735 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
495736 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
495737 2606.219 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
495738 2606.219 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
495739 2606.219 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
495740 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
495741 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
495742 2606.219 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
495750 2606.219 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
495751 2606.219 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
495752 2606.219 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
495753 2612.120 USER (SwingExceptionHandler.java:220) User requested sending bug report
495754 2612.124 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
[GW ContestRocketOR_v6.ork.txt](https://github.com/openrocket/openrocket/files/9991677/GW.ContestRocketOR_v6.ork.txt)

hcraigmiller, 2022-11-11:
Could not replicate using current unstable.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-11-11:
The chute error has already been fixed in #1726. The bug reported you posted is from a bug that has been fixed by #1732.

closed 2022-11-11 by SiboVG