openrocket/openrocket #1815: Component name changed after loading

benrussell11, 2022-11-11:
I've changed the component names by added a "S - " or "B - " for the sustainer and booster chutes.

[image]
[image]

File -> Save - Quit
File -> Open recent

Updates to the component names not saved
Sustainer
[image]
Booster
[image]

Design file attached
[GW ContestRocketOR_v7.ork.txt](https://github.com/openrocket/openrocket/files/9992781/GW.ContestRocketOR_v7.ork.txt)

hcraigmiller, 2022-11-11:
Behavior replicated in current unstable using parachutes from the parts list, but not standard (custom) parachutes (such as the A simple model rocket parachute). If you change the _A simple model rocket_ parachute to a parachute from the parts list, change its name, then save and reopen, the name reverts to the parts list name.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-11-11:
The issue is that the parachutes you renamed are preset parachutes. Currently, OpenRocket just always loads the preset component name. It should be possible to have custom names for preset components, without clearing the preset.

PR #1816 (2022-11-15): [#1815] Don't let preset loading override component name on file import
This PR fixes #1815.
closed 2022-11-15 by SiboVG